commaai/openpilot #34012: LKAS fault: Restart car to engage | Toyota hybrid 2022

sbr5136014, 2024-11-13:
### Describe the bug

I installed camera 3 with the open pilot master branch it's connecting to the car 
but after a minute I got this message "LKAS fault: Restart car to engage" 
(after the comma 3 is turned on, I can't change the lane assistance settings on the car) any help!
[image000000 (2)](https://private-user-images.githubusercontent.com/79987128/385373552-9ddf68e4-87e4-4512-b2de-cdcd7a641de2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NzQ2MzcsIm5iZiI6MTczMTQ3NDMzNywicGF0aCI6Ii83OTk4NzEyOC8zODUzNzM1NTItOWRkZjY4ZTQtODdlNC00NTEyLWIyZGUtY2RjZDdhNjQxZGUyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEzVDA1MDUzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5ODFjYzc4MTQ1YzAwODNmYzgwMDcxOTkwZjI4ZjQwZDQzODVkOTZlYjAyN2FkNDZlZmQ2NzRiZWUxYjVkMDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.IuboFBo8Oh5fcqr7F3oKWjivvdZJuOec6KRIRxdhowI)

### Which car does this affect?

Toyota sienna hybrid 2022

### Provide a route where the issue occurs

11/12/2024

### openpilot version

0.9.7

jyoung8607, 2024-11-13:
Did you take the necessary steps to recover the CAN security key? That's the error you'll see if you haven't done it. That's why the car isn't on the officially supported plug-and-play list.

Please join the comma Discord server and look at the pinned messages in #toyota-security for everything you need to know.

Side note: it's a bug that we don't communicate this well, you're supposed to be seeing a more informative message, but I haven't had a chance to investigate why the event isn't doing what I intended. You still need to do the same thing, we just suck at communicating it right now. 

https://github.com/commaai/openpilot/blob/96f6e186c4d07f5dff65b71122699bc8138af49b/selfdrive/selfdrived/events.py#L385-L389

jyoung8607, 2024-11-13:
The original end user can fix themselves with a bit of RTFM, but I'm leaving this issue open to track the Event bug.

sshane, 2024-11-15:
@jyoung8607 can you migrate issues? should we move to opendbc or close?

jyoung8607, 2024-11-15:
I don't have the mojo to migrate anymore, but we shouldn't need to. The events are handled in controls or car_specific here on the openpilot side. You could retitle it to "SecOC missing key event is broken".

sshane, 2024-11-15:
Need to do something like this:
https://github.com/commaai/openpilot/blob/87bc80d35145686903ccb499cd4fe4092054a45e/selfdrive/selfdrived/selfdrived.py#L131-L132

sshane, 2025-05-13:
Is this still an issue?