commaai/openpilot #34026: logs sometimes don't upload for routes with CAN network breakage

jyoung8607, 2024-11-14:
### Describe the bug

Have noticed several times, working with people on new car ports or custom harnesses, that I often have trouble getting uploaded logs when there's CAN errors.

I haven't tried to investigate. I can't say if logs aren't written, or aren't uploaded, or aren't accepted by the cloud.

I'm not sure when it started, but I would estimate several weeks ago.

### Provide a route where the issue occurs

05b94826f7ced550/00000003--7113599579

### openpilot version

c383ea866671264f883054964511514effb12a5f (today's master-ci)

sshane, 2024-11-21:
This route never had its qcams uploaded: a2bddce0b6747e10/000004d2--96024a48ab

[image]

adeebshihadeh, 2024-11-21:
qcams don't upload by default

sshane, 2024-11-21:
You can see I visited the connect page to request the rlogs, so the qcams should have uploaded. They did eventually upload.

closed 2024-11-25 by adeebshihadeh