commaai/openpilot #34041: comma three: get ADB working over USB

adeebshihadeh, 2024-11-16:
I used adb a bunch while doing the ISP debayer, and it's noticably snappier than SSH! Already works over the network using https://github.com/commaai/openpilot/blob/master/selfdrive/debug/adb.sh

Let's treat it the same as SSH:
* add a toggle to the openpilot settings UI for enabling it (default off)
* enabled if openpilot isn't installed yet https://github.com/commaai/agnos-builder/blob/master/userspace/usr/comma/comma.sh#L28

adeebshihadeh, 2024-11-25:
These are from a working Ubuntu image with adb up over usb. Not sure it's enough, but should at least point in the right direction.

[adbd.txt](https://github.com/user-attachments/files/17897074/adbd.txt)
[usb.txt](https://github.com/user-attachments/files/17897075/usb.txt)

amoghmunikote, 2024-11-25:
I think that might be enough for me to implement it.

adeebshihadeh, 2024-12-24:
Here's another internal script we use for provisioning that sets up gadget mode if that helps
[setup_gadget_mode.txt](https://github.com/user-attachments/files/18241832/setup_gadget_mode.txt)

amoghmunikote, 2024-12-26:
I'll try something with that

closed 2025-01-28 by adeebshihadeh