commaai/openpilot #34042: cabana: missing video

royjr, 2024-11-18:
### Describe the bug

[image]

### OS Version

macOS 15.1

### openpilot version or commit

master

royjr, 2024-12-09:
[image]
[image]
Seems to only affect road camera

deanlee, 2024-12-10:
Could you confirm if the route you’re working with includes road camera data?
Can you view the road camera by running `cabana --demo` with the demo route?

royjr, 2024-12-10:
None of the cameras work for the demo route in replay.

All but the road camera works in cabana 
`./cabana --demo --qcam --ecam --dcam`

If I swap file names f <-> e, f does show up instead of e, so its not a file/encoding issue

[image]
[image]
[image]
[image]

royjr, 2024-12-10:
Confirmed flags are being set correctly and files urls are good

[image]

deanlee, 2024-12-10:
It looks like this is a port conflict issue. VisionIPC on macOS uses fixed ports for streaming camera data over ZeroMQ:

- 9000 for the road camera,
- 9001 for the driver camera,
- 9002 for the wide road camera.

Please check if port 9000 is currently occupied by another application on your system. You can verify this by running `lsof -i :9000`. If it's in use, you’ll need to free it to allow VisionIPC to function properly.

royjr, 2024-12-10:
That worked, thanks!

deanlee, 2024-12-10:
Glad to hear that worked!

royjr, 2024-12-11:
Video also freezes and doesn't recover until app is restarted. (let me know if this needs a new issue)
- closing the stream and reopening using the menu doesn't help.
- only freezes when seeking (both using the seek bar and using the signals)
- happens on both local and remote routes

```
Starting listener for: camerad
Error sending a packet for decoding: -35
Error sending a packet for decoding: -35
Error sending a packet for decoding: -35
camera[0] failed to get frame: 905
Error sending a packet for decoding: -35
...
Error sending a packet for decoding: -35
camera[0] failed to get frame: 906
Error sending a packet for decoding: -35
...
Error sending a packet for decoding: -35
camera[0] failed to get frame: 907
Error sending a packet for decoding: -35
...
Error sending a packet for decoding: -35
camera[0] failed to get frame: 908
Error sending a packet for decoding: -35
```

royjr, 2024-12-11:
[image]
[image]

deanlee, 2024-12-11:
What's your ffmpeg version? Use `ffmpeg -version` to check.

royjr, 2024-12-11:
```
ffmpeg version 7.1 Copyright (c) 2000-2024 the FFmpeg developers
built with Apple clang version 16.0.0 (clang-1600.0.26.4)
configuration: --prefix=/opt/homebrew/Cellar/ffmpeg/7.1_3 --enable-shared --enable-pthreads --enable-version3 --cc=clang --host-cflags= --host-ldflags='-Wl,-ld_classic' --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libaribb24 --enable-libbluray --enable-libdav1d --enable-libharfbuzz --enable-libjxl --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librist --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-videotoolbox --enable-audiotoolbox --enable-neon
libavutil      59. 39.100 / 59. 39.100
libavcodec     61. 19.100 / 61. 19.100
libavformat    61.  7.100 / 61.  7.100
libavdevice    61.  3.100 / 61.  3.100
libavfilter    10.  4.100 / 10.  4.100
libswscale      8.  3.100 /  8.  3.100
libswresample   5.  3.100 /  5.  3.100
libpostproc    58.  3.100 / 58.  3.100
```

deanlee, 2024-12-11:
Could you help test if PR https://github.com/commaai/openpilot/pull/34205 resolves the issue?

royjr, 2024-12-11:
Its better but still freezes
[image]
[image]

royjr, 2024-12-11:
It does unfreeze.. at the start of the next segment

deanlee, 2024-12-11:
Have other users encountered the same issue on macOS?

royjr, 2024-12-11:
not sure, but its defenitely better, and no freezes using one camera

closed 2025-02-09 by adeebshihadeh