commaai/openpilot #34049: Setup failure on macOS (Apple Silicon)

sunnyhaibin, 2024-11-18:
### Describe the bug

Ran `tools/mac_setup.sh` on commit https://github.com/commaai/openpilot/commit/4576f7f1547e29cc50f97b6674b950ef03e39016

```
% ./tools/mac_setup.sh
Using cppcheck
Using git-lfs
Using zlib
Using capnp
Using coreutils
Using eigen
Using ffmpeg
Using glfw
Using libarchive
Using libusb
Using libtool
Using llvm
Using openssl@3.0
Using qt@5
Using zeromq
Installing gcc-arm-embedded
==> Downloading https://developer.arm.com/-/media/Files/downloads/gnu/13.3.rel1/binrel/arm-gnu-toolchain-13.3.rel1-darwin-arm64-arm-none-eabi.pkg
curl: (22) The requested URL returned error: 404
Error: Download failed on Cask 'gcc-arm-embedded' with message: Download failed: https://developer.arm.com/-/media/Files/downloads/gnu/13.3.rel1/binrel/arm-gnu-toolchain-13.3.rel1-darwin-arm64-arm-none-eabi.pkg
Installing gcc-arm-embedded has failed!
Using portaudio
Using gcc@13
Homebrew Bundle failed! 1 Brewfile dependency failed to install.
```

### OS Version

macOS Sequoia 15.1

maxime-desroches, 2024-11-18:
All the links for the Arm GNU Toolchain 13.3 from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads return 404 now. We might want to switch to 13.2 until they fix this

sunnyhaibin, 2024-11-19:
Installed 13.2 for the time being and successfully built `openpilot`. For any future searchers, here are the steps to install `gcc-arm-toolchain` with `13.2.rel1` (commit used with [`13.2.rel1`](https://github.com/Homebrew/homebrew-cask/blob/0d591515ad2a153113e0316d31fe06915fb07ac0/Casks/g/gcc-arm-embedded.rb) at [`Homebrew/homebrew-cask`](https://github.com/Homebrew/homebrew-cask)):
```zsh
cd ~/

curl -L https://github.com/Homebrew/homebrew-cask/blob/0d591515ad2a153113e0316d31fe06915fb07ac0/Casks/g/gcc-arm-embedded.rb > gcc-arm-embedded.rb

brew install --cask ~/gcc-arm-embedded.rb
```

maxime-desroches, 2024-11-19:
Links are working again

closed 2024-11-19 by maxime-desroches