commaai/openpilot #34064: AGNOS 11.2 comma3 sound stopped working

AlexandreSato, 2024-11-19:
### Describe the bug

After the update the sound stopped working.

### Provide a route where the issue occurs

075b133b6181e058

### openpilot version

293c3fc57fc7de70a43a257a188c433a66d909ff

### Additional info

I reverted the commit and the sound started working again.

adeebshihadeh, 2024-11-21:
How did you flash it? Sounds like the kernel and system aren't matched, which can happen if you flashed manually.

AlexandreSato, 2024-11-21:
> How did you flash it? Sounds like the kernel and system aren't matched, which can happen if you flashed manually.

I also tried via QDL and there is still no sound

adeebshihadeh, 2024-11-22:
@robbederks seems it's our new kernel driver: https://github.com/commaai/agnos-kernel-sdm845/pull/63

AlexandreSato, 2024-11-22:
I flashed it again using flash.comma.ai and then let updaterd update automatically and it still had no sound: 075b133b6181e058/00000001--5e00c93cc4/0

robbederks, 2024-11-22:
> @robbederks seems it's our new kernel driver: https://github.com/commaai/agnos-kernel-sdm845/pull/63

Interesting, tested it on a 3X and that was fine

adeebshihadeh, 2024-11-26:
Gets even more interesting. Seems like this needs a cold boot to take effect...

robbederks, 2024-11-26:
Maybe `amplifier.py` doesn't set all the required registers on the codec and it was relying on the kernel driver to do some initial setup?
Or do you mean the other way around where it only works after a cold boot?

closed 2024-12-05 by robbederks