commaai/openpilot #34066: Disengage lateral control gradually over 1 second when brake is pressed on a curved road to allow for the driver to grab the handle

calvinpark, 2024-11-20:
### Scenario
* Driver is paying attention and is ready to grab the handle and press the brake in case of an emergency.
* Car is driving on a freeway on a road that's curved to the right with OP engaged.
* Car in front of of the driver's car suddenly brakes hard.
* Driver press the brake and grabs the handle. OP disengages both long and lat controls.

### Behavior with room for improvements
* The driver will instinctively press the brake and grab the handle at the same time. However, pressing the brake and grabbing the handle doesn't happen at the same exact time.
* If the driver grabs the handle first and then presses the brake a split second later, the car's lateral trajectory is unchanged (since the handle is grabbed while turned) while the car decelerates.
* If the driver presses the brake and then grabs the handle a split second later, the car's lateral trajectory is changed because OP released the lateral control while driving on a curved road.

### Two problems
1. The car will swing to the left for a split second and then the driver will correct it back to right. This creates a swerve which makes the emergency situation more dangerous.
2. The handle will start to turn CCW (because the car wants to go straight) which makes it more difficult to grab. A stationary handle is easier to grab than a spinning handle.

### Suggestion
When brake is pressed while going on a curved road,
1. Disengage long control immediately.
2. Disengage lat control if the handle is grabbed.
3. Release lat control slowly over 1 second, as opposed to immediately disengaging.
4. If the handle is grabbed before the end of 1 second, immediately disengage as soon as the handle is grabbed.

### Rationale
1. This gradual release gives the driver 1 extra second to grab the handle before the lateral control is completely disengaged.
2. The handle will be turning slower due to the gradual release, making it easier to grab.
3. There are no situations where it is safer for OP to disengage lateral control immediately while braking on a curved road.
5. If the driver wishes to turn the handle to avoid running into the car ahead, OP would disengage as soon as the handle is grabbed and thus the driver's action is not impeded.
6. This is a very short action in a specific situation, and thus the drivers wouldn't even notice that there was a change.
7. Lateral control is fully disengaged after 1 second anyway. This is not a version of Always-On-Lateral feature.

### Alternatives
1. The gradual release can be engaged only when the brake is fully engaged (to signal emergency).
2. It can also be engaged above a certain speed beyond a certain angle.
3. This behavior can be optional.

adeebshihadeh, 2024-11-20:
We'd be open to checking out a PR that implements this, but I'm going to close since it's not on our immediate roadmap.

closed 2024-11-20 by adeebshihadeh

gregjhogan, 2024-11-21:
FYI, this can be implemented for many Hyundai vehicles without any panda safety changes:
https://github.com/commaai/openpilot/pull/20511/files

calvinpark, 2024-11-22:
@gregjhogan Nice, it's the same idea. But it seems like Comma is (was?) against idea. I think it's a good idea, but all I can do is to bring it up again.