commaai/openpilot #34074: No auth.json file

renyusang, 2024-11-21:
### Describe the bug

I run `tools/replay/replay --demo` on PC docker, terminal shows following errors
```bash
loading route a2a0ccea32023010|2023-07-27--13-01-19
Error parsing auth.json unexpected end of input
Retrying 1/3
Error parsing auth.json unexpected end of input
Retrying 2/3
Error parsing auth.json unexpected end of input
Retrying 3/3
failed to load route a2a0ccea32023010|2023-07-27--13-01-19
```
I find no auth.json in the docker container, how can I get an auth.json?
Or is there other way to run the sample `Turn the speed blue`
Thanks

docker image `ghcr.io/commaai/openpilot-base:latest`
docker pull date: 20241121 

### OS Version

Ubuntu 2004

### openpilot version or commit

87bc80d35145686903ccb499cd4fe4092054a45e

maxime-desroches, 2024-11-21:
That file is normally put in `~/.comma` after running `tools/lib/auth.py`
Since you are on ubuntu 20.04, you should just run everything natively without docker

renyusang, 2024-11-23:
> That file is normally put in `~/.comma` after running `tools/lib/auth.py` Since you are on ubuntu 20.04, you should just run everything natively without docker

Thanks for your reply, this Issue closed

closed 2024-11-23 by renyusang