commaai/openpilot #34079: Possible athenad memory leak

sshane, 2024-11-21:
It was using 15% of memory, causing low memory events. I had not rebooted the device in a few days iirc and had uploaded many rlogs.

I tried to attach with memray, but instead the process just crashed so I didn't get any information about this. @deanlee could this be similar to the leak you found in athena previously?

deanlee, 2024-11-22:
It's hard to identify the exact issue, as the code is unclear and difficult to understand. However, multiple factors could be causing the resource leak. One possibility is that the athenad's implementation might cause garbage collector to accumulate objects over time, leading to memory issues during long runs.
additionally, thread race conditions related to queue and other global variable usage, as well as bad practices like not calling task_done() after queue.get(), could also contribute to issues.

deanlee, 2024-11-23:
The following issues and fixes address potential resource leaks and long-term accumulation risks:
1. **Unclosed WebSocket:** Not closing the WebSocket after a main loop restart can cause resource leaks by keeping connections open. https://github.com/commaai/openpilot/pull/34085
2.  **Thread Safety in Upload Handling:** Improper synchronization can lead to race conditions, potentially causing threads to terminate and forcing the main loop to restart. This may result in invalid or inconsistent data being stored in the upload queue, leading to data corruption or queue bloat.  https://github.com/commaai/openpilot/pull/34084
3. **Memory Bloat in cur_upload_items[tid]**: The `cur_upload_items` dictionary isn't properly cleared of invalid or duplicate entries. If threads restart with new IDs, the dictionary can grow indefinitely, causing memory bloat over time.

sshane, 2024-12-05:
This is still happening after uploading files

sshane, 2024-12-05:
I see this after uploading rlogs:

```
lock                            120       +47
cell                            828       +47
builtin_function_or_method     1685       +43
list                           1776       +38
tuple                          4715       +30
ReferenceType                  2736       +30
Event                            75       +15
socket                           24       +15
Condition                        54       +14
deque                            62       +14
method                          109       +13
function                       7974       +13
SSLContext                       14        +8
WebSocket                        13        +8
sock_opt                         13        +8
frame_buffer                     13        +8
continuous_frame                 13        +8
SplitResult                      14        +8
SSLSocket                        14        +8
_SSLSocket                       14        +8
```

deanlee, 2024-12-06:
The issue may not just be with the upload process—WebSocket instance growth could indicate deeper resource management problems.

deanlee, 2024-12-06:
@sshane: Do we really need the `ws_manage` thread? It doesn’t handle exceptions, and changing socket options while the WebSocket is sending or receiving could cause race conditions or undefined behavior. If socket options need to be set, a better approach would be to close the connection, reconnect, and configure the socket before restarting `ws_recv`/`ws_send` threads.

Also, storing sock = ws.sock at the start of the thread and assuming it's valid throughout the loop may not be safe.

deanlee, 2024-12-06:
The current `athenad` implementation lacks consideration for concurrent data access safety in a multi-threaded environment. Besides https://github.com/commaai/openpilot/pull/34084, which fixes race conditions for global variables, the UploadQueueCache class also has multi-threading issues. UploadQueueCache may be accessed by multiple threads simultaneously, but it is not thread-safe.

deanlee, 2024-12-07:
@sshane : The athenad implementation does not maintain the order between received commands and their corresponding responses. Responses may be sent in a different order than the commands were received. Is this behavior expected and by design, or should it be addressed as a issue?

jakethesnake420, 2024-12-13:
> @sshane: The current implementation does not preserve the order between received commands and their responses. Responses may be sent out of order.
> 
> Commands are placed in the `recv_queue` and processed by multiple `jsonrpc_handler` threads, with responses added to the `send_queue` after handling. Due to the multithreaded nature, the order of responses may differ from the order of received commands.
> 
> Is this behavior by design, or should it be considered an issue?

The order of responses does not matter since the JsonRPC spec handles this. The 'requester' sends a request with a unique ID which could just be a ns timestamp for example. When the 'responder' (athena in this case) responds, it will also send the ID in the payload. https://www.jsonrpc.org/specification#response_object

sshane, 2025-01-17:
Can't seem to reproduce this by re-uploading a large number of rlogs

adeebshihadeh, 2025-02-07:
One instance onroad:
```
21d8936b8bd24158/000000fa--4974fc8223/0
```

sshane, 2025-02-07:
Removing the zstd compression drastically reduced the usage after uploading files for me.

sshane, 2025-02-20:
Memory usage of athenad is only 1.4% after uploading a few routes, think this was something zstd was doing

closed 2025-02-20 by sshane