commaai/openpilot #34114: KIA EV6 2025 Steering not working

taylor224, 2024-11-27:
### Describe the bug

Car model : KIA EV6 2025 (facelifted model)
Comma model : Comma 3x
Car harness : Hyundai P

Hello, I applied car model EV6 22-24 (without HDA2), but steering control not working.
Comma 3x shows cyan square border and steering wheel icon not showing on my cluster.
I think it occur because it is not registered fingerprint. How can I help to register this car to Openpilot?
Thank you.

### Which car does this affect?

KIA EV6 2025 (without HDA2)

### Provide a route where the issue occurs

84711a8268bf3b9a/00000007--9fa77cf42e

### openpilot version

latest production release

taylor224, 2024-11-27:
It similar with this issue
https://github.com/commaai/openpilot/pull/30931

taylor224, 2024-11-27:
Hello, I read some docs and uploaded PR to opendbc.
I noticed my car has HDA2.

https://github.com/commaai/opendbc/pull/1534

sunnyhaibin, 2025-03-12:
2025 EV6 Facelift is angle control and requires a car port.

closed 2025-08-01 by adeebshihadeh