commaai/openpilot #34120: Phantom touch on C3X

haraschax, 2024-11-27:
### Describe the bug

Sidebar was opening and closing randomly all drive

https://github.com/user-attachments/assets/ad15d31e-831d-47ef-99f0-28c91d2e089f

### Provide a route where the issue occurs

57048cfce01d9625/00000001--1931938d1c

### openpilot version

master-branch

deanlee, 2024-11-27:
The sidebar only appears when there is a click event or when transitioning from "onroad" to "offroad," so this issue doesn’t seem to be caused by the OP itself. If the hardware is not at fault, it could be due to faulty or misconfigured input devices. such as malfunctioning or overly sensitive settings.

nelsonjchen, 2024-12-10:
This issue is confusing. At least in discord, we would say to contact support, probably a hardware issue, but you're comma staff... 

Is this a software issue?

ErichMoraga, 2024-12-13:
Normally it's a digitizer (hardware) issue.
ie. Swap out the screen.

closed 2024-12-13 by adeebshihadeh