commaai/openpilot #34145: Unable to Install master branch

morrislee, 2024-12-03:
### Describe the bug

Master branch or anything based on master recently has not been able to complete install. Looks like it fails at Agnos both using SSH command or regulat update UI button

screenshot shows some sort of size chunk issue?

[image]
[image]

### Provide a route where the issue occurs

dc7716b32bf25574

### openpilot version

0.9.8

jakethesnake420, 2024-12-03:
I have the same issue.

nelsonjchen, 2024-12-03:
Possibly related to https://github.com/commaai/openpilot/issues/34138 ?

morrislee, 2024-12-03:
> Possibly related to #34138 ?

looks like the same concern. can be replicated sinply try to download the system image off http link

morrislee, 2024-12-04:
tried installing it just now, started working, hopefully it stays working when I flip back and forth on forks

nelsonjchen, 2024-12-04:
I think this is a bit out of comma's control and a bit in.

https://github.com/commaai/openpilot/blob/a22d6cd0d31b19b0d218812008b0a18011f286bf/system/hardware/tici/agnos.py#L21-L54

https://github.com/commaai/openpilot/blob/a22d6cd0d31b19b0d218812008b0a18011f286bf/system/hardware/tici/agnos.py#L165-L187

@jakethesnake420 's example with `wget` even show some resilience on `wget`'s part with auto-resumes. However, the agnos downloader is a bit basic and assumes Azure's CDN isn't closing connections prematurely.

morrislee, 2024-12-04:
Would it be a good idea if it is a "submodule" and simply caches the image file to the disk via github, and then it writes to the disk then deletes it later? Or this is too much overhead for github terms and conditons?

jakethesnake420, 2024-12-04:
The correct solution is to move this issue onto Microsoft Azure team if it's an issue with their service or fix the agnos staging code if there is an issue with that.

adeebshihadeh, 2024-12-05:
6c02d5c3f7fe44d60707fbc3e647b982300d0c0f

closed 2024-12-05 by adeebshihadeh