commaai/openpilot #34197: camerad fails to start when trying to use webcam with openpilot in WSL2 Ubuntu

MikeBusuttil, 2024-12-09:
### Describe the bug

Following the steps in [the webcam readme](https://github.com/commaai/openpilot/blob/master/tools/webcam/) and running `NOSENSOR=1 USE_WEBCAM=1 system/manager/manager.py` appears to work.  UI loads and console is all green in console:
```
(openpilot) mike@MIKEZLAP:~/openpilot$ NOSENSOR=1 USE_WEBCAM=1 system/manager/manager.py
system/loggerd/bootlog.cc: bootlog to /home/mike/.comma/media/0/realdata/boot/00000001--cad91c878d
missing public key: /home/mike/.comma/persist/comma/id_rsa.pub
selfdrive/ui/qt/network/wifi_manager.cc: DBus call error: "The name org.freedesktop.NetworkManager was not provided by any .service files"
selfdrive/ui/qt/network/wifi_manager.cc: DBus call error: "The name org.freedesktop.NetworkManager was not provided by any .service files"
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
logmessaged ui pandad deleter hardwared statsd
logmessaged ui pandad deleter hardwared statsd
logmessaged ui pandad deleter hardwared statsd
```

but then results in the following when pressing the `Driver Camera` `Preview` button in the UI (red `camerad` and `dmonitoringmodeld`):

```
logmessaged ui pandad deleter hardwared statsd
logmessaged ui pandad deleter hardwared statsd
Process camerad:
Traceback (most recent call last):
  File "/home/mike/.local/share/uv/python/cpython-3.12.0-linux-x86_64-gnu/lib/python3.12/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/mike/.local/share/uv/python/cpython-3.12.0-linux-x86_64-gnu/lib/python3.12/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/mike/openpilot/openpilot/system/manager/process.py", line 56, in nativelauncher
    os.execvp(pargs[0], pargs)
  File "<frozen os>", line 580, in execvp
  File "<frozen os>", line 603, in _execvpe
FileNotFoundError: [Errno 2] No such file or directory
camerad logmessaged dmonitoringmodeld ui pandad deleter dmonitoringd hardwared statsd
selfdrive.modeld.dmonitoringmodeld: common/clutil.cc:79: cl_context cl_create_context(cl_device_id): Assertion `_ret&& err == CL_SUCCESS' failed.
camerad logmessaged dmonitoringmodeld ui pandad deleter dmonitoringd hardwared statsd
camerad logmessaged dmonitoringmodeld ui pandad deleter dmonitoringd hardwared statsd
camerad logmessaged dmonitoringmodeld ui pandad deleter dmonitoringd hardwared statsd
```

### OS Version

Ubuntu 24.04 (WSL2)

### openpilot version or commit

57fc4f79d16c132d526a66d603fd9100804569d1

### Additional info

- The [webcam readme](https://github.com/commaai/openpilot/blob/57fc4f79d16c132d526a66d603fd9100804569d1/tools/webcam/README.md?plain=1#L27) references `system/camerad/cameras/camera_webcam.cc` which no longer exists.

jyoung8607, 2024-12-10:
I would not expect this to work easily on WSL2, because host-attached USB devices are not passed through to WSL like they might be for traditional hypervisors like VMware Workstation. There's some workarounds with USB-over-IP, but setting it up is an extremely manual process that I'm not sure anyone has attempted with openpilot, and I'm not sure how well it would work with high-bandwidth devices like cameras.

All that said, I'm leaving this bug open because webcam does look legitimately broken on baremetal Ubuntu. I was able to reproduce the described issue. I instrumented process.py to tell me what it was executing:

```diff
diff --git a/system/manager/process.py b/system/manager/process.py
index 0e9c9b980..31a54a1b3 100644
--- a/system/manager/process.py
+++ b/system/manager/process.py
@@ -53,6 +53,7 @@ def nativelauncher(pargs: list[str], cwd: str, name: str) -> None:
 
   # exec the process
   os.chdir(cwd)
+  print(f"{cwd=} {pargs}")
   os.execvp(pargs[0], pargs)
```

Which obtained the following result:

```
(openpilot) jyoung@jy-workstation-ubuntu:~/openpilot$ USE_WEBCAM=1 NOSENSOR=1 system/manager/manager.py 
system/loggerd/bootlog.cc: bootlog to /home/jyoung/.comma/media/0/realdata/boot/00000008--b1121cf470
system/manager/process.py: needs update
missing public key: /home/jyoung/.comma/persist/comma/id_rsa.pub
cwd='/home/jyoung/openpilot/selfdrive/ui' ['./ui']
logmessaged ui pandad deleter hardwared statsd
logmessaged ui pandad deleter hardwared statsd
logmessaged ui pandad deleter hardwared statsd
logmessaged ui pandad deleter hardwared statsd
logmessaged ui pandad deleter hardwared statsd
logmessaged ui pandad deleter hardwared statsd
cwd='/home/jyoung/openpilot/system/camerad' ['./camerad']
Process camerad:
Traceback (most recent call last):
  File "/home/jyoung/.local/share/uv/python/cpython-3.12.0-linux-x86_64-gnu/lib/python3.12/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/jyoung/.local/share/uv/python/cpython-3.12.0-linux-x86_64-gnu/lib/python3.12/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/jyoung/openpilot/openpilot/system/manager/process.py", line 57, in nativelauncher
    os.execvp(pargs[0], pargs)
  File "<frozen os>", line 580, in execvp
  File "<frozen os>", line 603, in _execvpe
FileNotFoundError: [Errno 2] No such file or directory
cwd='/home/jyoung/openpilot/selfdrive/modeld' ['./dmonitoringmodeld']
camerad logmessaged dmonitoringmodeld ui pandad deleter dmonitoringd hardwared statsd
camerad logmessaged dmonitoringmodeld ui pandad deleter dmonitoringd hardwared statsd
```

The error comes from trying to start `system/camerad/camerad` which does not build on `larch64` (Linux on PC), we should be trying to launch `tools/webcam/camerad.py`. Or perhaps `start_camerad.sh`? It's not immediately clear to me how this is supposed to work. This might be fallout from some of the selfdrive -> system migrations a while back.

**TL;DR:** Don't use WSL2, but this is a real bug anyway.

MikeBusuttil, 2024-12-10:
Thanks @jyoung8607 for digging in!  Indeed getting USB cameras working on WSL2 was a big pain.  It involved both:
- a 3rd party Windows USB sharing software ([usbipd](https://github.com/dorssel/usbipd-win)) and
- rebuilding the WSL2 Linux Kernel (good instructions [here](https://www.youtube.com/watch?v=t_YnACEPmrM))

Keep me posted if you make any more progress, I'll time-box poking around at this and if I can't get it going by the evening I'll find another issue to pick up.

maoyuezxc, 2024-12-11:
I come across the same issue，which can be solved by running  tools/webcam/start_camerad.sh first.  In the bash script, the right number for the environment variables of each camera should be specified.

Pay attention to the problem  that the index  number of /dev/video  may change each time the camera is plugged.

jyoung8607, 2024-12-11:
I had noticed that, specifically the `BLOCK` export it does to prevent manager from starting the normal `camerad`, but hadn't had a chance to test.

@MikeBusuttil: Give that a try, and if you find a working path, feel free to file PRs against the README or anything else necessary to make this easier for new users.

MikeBusuttil, 2024-12-11:
Tried it on 2 machines with different failures.

## Ubuntu 24.04 on old Dell i5 Laptop:
```bash
(openpilot) willow@willow-Latitude-E5450:~/openpilot$ tools/webcam/start_camerad.sh 
Traceback (most recent call last):
  File "/home/willow/openpilot/tools/webcam/camerad.py", line 74, in <module>
    main()
  File "/home/willow/openpilot/tools/webcam/camerad.py", line 69, in main
    camerad = Camerad()
              ^^^^^^^^^
  File "/home/willow/openpilot/tools/webcam/camerad.py", line 33, in __init__
    self.vipc_server.create_buffers(c.stream_type, 20, False, cam.W, cam.H)
  File "msgq_repo/msgq/visionipc/visionipc_pyx.pyx", line 65, in msgq.visionipc.visionipc_pyx.VisionIpcServer.create_buffers
    def create_buffers(self, VisionStreamType tp, size_t num_buffers, size_t width, size_t height):
TypeError: create_buffers() takes exactly 4 positional arguments (5 given)
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
tools/webcam/start_camerad.sh: line 14:  3803 Aborted                 (core dumped) $DIR/camerad.py
```
which is what I'd expect given the [create_buffers interface](https://github.com/commaai/msgq/blob/434ed2312c980b5504a4a75001d04c3ecbddf93f/msgq/visionipc/visionipc_server.cc#L41)
```cpp
void VisionIpcServer::create_buffers(VisionStreamType type, size_t num_buffers, size_t width, size_t height){
```
so I removed the 3rd argument from the [offending line](https://github.com/commaai/openpilot/blob/02976db4729207a8600e77f0d9dd6a0fd7b245c2/tools/webcam/camerad.py#L30) to make it now
```py
self.vipc_server.create_buffers(c.stream_type, 20, cam.W, cam.H)
```
which results in
```bash
(openpilot) willow@willow-Latitude-E5450:~/openpilot$ tools/webcam/start_camerad.sh 
Traceback (most recent call last):
  File "/home/willow/openpilot/tools/webcam/camerad.py", line 74, in <module>
    main()
  File "/home/willow/openpilot/tools/webcam/camerad.py", line 69, in main
    camerad = Camerad()
              ^^^^^^^^^
  File "/home/willow/openpilot/tools/webcam/camerad.py", line 28, in __init__
    cam = Camera(c.msg_name, c.stream_type, c.cam_id)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/willow/openpilot/openpilot/tools/webcam/camera.py", line 13, in __init__
    self.container = av.open("/dev/video0")
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "av/container/core.pyx", line 420, in av.container.core.open
  File "av/container/core.pyx", line 266, in av.container.core.Container.__cinit__
  File "av/container/core.pyx", line 286, in av.container.core.Container.err_check
  File "av/error.pyx", line 326, in av.error.err_check
av.error.OSError: [Errno 16] Device or resource busy: '/dev/video0'
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
tools/webcam/start_camerad.sh: line 14:  3586 Aborted                 (core dumped) $DIR/camerad.py
```
But `/dev/video0` of course works just fine elsewhere (ie. in `cheese`) so I'm stuck there for now.

## WSL Ubuntu 24.04 on Dell XPS 17

I can't get `dmonitoringmodeld` to run
```
selfdrive.modeld.dmonitoringmodeld: common/clutil.cc:79: cl_context cl_create_context(cl_device_id): Assertion `_ret&& err == CL_SUCCESS' failed.
```
despite having my environment setup properly (I thought)
[image]
leaving me stuck on that machine as well.

Will continue poking at 🤞

MikeBusuttil, 2024-12-12:
Good stuff @jyoung8607 💪 

Here's the draft PR to squash this bug: #34215

Will ping you when it's ready to rock

MikeBusuttil, 2024-12-12:
We've made a lot of progress here but it looks like there may be more roadblocks on the path to full webcam support.  It looks like the C3X camera resolution is hard-coded into `dmonitoringmodeld`

[image]

MikeBusuttil, 2024-12-22:
It's working!! 😃
[image]
great work whoever got the final fix in last week 💪 (Harold, Adeeb, Yassine, Maxine, ZwX ?)

closed 2024-12-22 by MikeBusuttil

Willian7004, 2025-04-10:
I got a similar issue when running the demo with `tools/replay/replay --demo` on WSL2 . The demo works well but I got a similar error  when starting the ui.
```
william@DESKTOP-G98V0AL:~/openpilot$ selfdrive/ui/ui
selfdrive/ui/qt/network/wifi_manager.cc: DBus call error: "The name org.freedesktop.NetworkManager was not provided by any .service files"
selfdrive/ui/qt/network/wifi_manager.cc: DBus call error: "The name org.freedesktop.NetworkManager was not provided by any .service files"
Segmentation fault
```