commaai/openpilot #34216: WSL2 Ubuntu webcam support broken

MikeBusuttil, 2024-12-12:
### Describe the bug

Note: Depends on #34215

When running openpilot with webcam support and previewing the driver camera, the camera fails to stream (see [bottom image](https://github.com/user-attachments/assets/034e96ec-001c-4c54-b3cb-5053668b9180)).

We would expect to see `opening driverCameraState at /dev/video2` and `Starting listener for: camerad` in the console but instead it appears to get stuck at `opening roadCameraState at /dev/video0`

### Reproduction steps:
```
git clone git@github.com:MikeBusuttil/openpilot.git
git checkout dehack-webcam
ROAD_CAM=4 DRIVER_CAM=2 system/manager/manager.py
```
Note: your camera numbers may differ from mine

### Here it is working as expected on native Ubuntu 24.04
[image]

### Here it is failing to work onUbuntu 24.04 WSL2
[image]

### OS Version

Ubuntu 24.04 (WSL2)

jyoung8607, 2024-12-12:
This isn't a bug. There was never "support" for webcam mode on WSL2.

closed 2024-12-12 by jyoung8607

MikeBusuttil, 2024-12-12:
[image]