commaai/openpilot #34255: Tools/Joystick now working properly

vcs-vct, 2024-12-16:
I checked out the master branch of openpilot. My plan was to use a joystick to control a 2021 Kia Telluride. I calibrated and ran openpilot on public road. I used Comma 3X and everything on public road with openpilot was fine.

I followed exact procedures in README to run joystick with keyboard.

"The car must be off, and openpilot must be offroad before starting joystick_control.

Using a keyboard
SSH into your comma device and start joystick_control with the following command:

tools/joystick/joystick_control.py --keyboard
The available buttons and axes will print showing their key mappings. In general, the WASD keys control gas and brakes and steering torque in 5% increments."

Once I started the vehicle, I got the following error
"
Killing old publisher: testJoystick
Exception in thread Thread-1 (send_thread):
Traceback (most recent call last):
  File "/usr/local/uv/python/cpython-3.12.3-linux-aarch64-gnu/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/local/uv/python/cpython-3.12.3-linux-aarch64-gnu/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/data/openpilot/tools/joystick/joystick_control.py", line 107, in send_thread
    pm.send('testJoystick', joystick_msg)
  File "/data/pythonpath/cereal/messaging/__init__.py", line 258, in send
    self.sock[s].send(dat)
  File "msgq_repo/msgq/ipc_pyx.pyx", line 242, in msgq.ipc_pyx.PubSocket.send
    raise MultiplePublishersError
msgq.ipc_pyx.MultiplePublishersError: Messaging failure : Address already in use
"

adeebshihadeh, 2025-01-20:
You can just use it from the developer panel in the settings now, no SSH necessary.

closed 2025-01-20 by adeebshihadeh

BBBmau, 2025-02-10:
> You can just use it from the developer panel in the settings now, no SSH necessary.

you still need to be able to run `joystick_control.py` IF you're trying to use keyboard which the author was attempting to do (and so am I)

There hasn't been much luck with keyboard method and just seems to be broken.

TorgenH, 2025-07-30:
Any updates on this topic? I ran into the same issue.