commaai/openpilot #34262: Cruise Fault when powering the car for the 1st time in the day

renatosc, 2024-12-16:
### Describe the bug

I have been using comma 3 for 1.5 years now and in the last few months I started to get the "Cruise fault" error a lot (I don't think I ever had that error on the first year using Comma). 

I was able to identify that the error happens when I am powering on my car usually in the morning and the Comma was totally powered off. 

So my car powers on while the Comma is still loading and then Comma indicates that that Car is offline and only after Comma indicates the Car is online I can then power off and on my car and the error goes away (If I try to power off/on the car while Comma indicates the Car is offline, it will still throw the Cruise fault error).

### Which car does this affect?

Honda Oyssey 2019

### Provide a route where the issue occurs

n/a

### openpilot version

0.9.7

sshane, 2024-12-16:
Can you provide a route where this happens?

1okko, 2024-12-17:
There should be many people encountering the same problem，like me