commaai/openpilot #34271: Hotspot does not work the first time you enable it

sshane, 2024-12-17:
This is because the time to add the connection takes ~2s, so our DBUS `call` method times out in 0.1s and doesn't save it as a connection. We can fix this by first saving it to memory `save no`, then write it out later by using the `Save` method. This does take 0.2s though, so we'll want to raise the DBUS timeout a bit.

@deanlee this should be a pretty simple fix if you want to do it!

sshane, 2024-12-17:
Actually an issue for normally connecting to APs too

sshane, 2024-12-17:
That sounds like a separate change we can evaluate later. This should be a ~4 line change

deanlee, 2024-12-17:
The two-step approach of calling AddConnectionUnsaved, followed by the Save() method on the returned connection object, can still result in timeout issues, particularly depending on the configured timeout settings.
A better solution is to use asyncCall to handle these operations asynchronously. This approach ensures the UI remains responsive, as it doesn’t block while waiting for the operation to complete. By doing so, we can avoid timeouts while still successfully saving the connection: https://github.com/commaai/openpilot/pull/34273

sshane, 2024-12-17:
cant repro the general connection issue, only tethering. fixed by https://github.com/commaai/openpilot/pull/34274

closed 2024-12-17 by sshane