commaai/openpilot #34276: AGNOS 11.4/11.3 slower NetworkManager calls

sshane, 2024-12-17:
The 11.3/11.4 AGNOS update causes many NetworkManager calls to be much slower now, so there may be ui networking regressions. For example: once immediate, forgetting connections will now always take 3s and time out. Going through the WifiManager to see if there's any user experience regressions

sshane, 2024-12-17:
https://github.com/commaai/openpilot/pull/34274

sshane, 2024-12-17:
https://github.com/commaai/openpilot/pull/34275

jyoung8607, 2025-01-12:
Resolution of commaai/agnos-builder#423 may resolve this as well.

adeebshihadeh, 2025-01-18:
AGNOS 11.5 seems fine to me (https://github.com/commaai/openpilot/pull/34413).

Can you verify? I'll wait to merge it until you do.

sshane, 2025-01-22:
Did you test deleting and creating connections? Going to test that now

sshane, 2025-01-22:
@adeebshihadeh It still takes 3-4s to delete/create a connection, you noticed earlier than systemd is reloading something, and that still seems to be the case:

```
Jan 22 03:08:44 comma-1a8caafe systemd[1]: Reloading requested from client PID 50769 ('systemctl') (unit NetworkManager.service)...                                              
Jan 22 03:08:44 comma-1a8caafe systemd[1]: Reloading...                                                                                                                          
Jan 22 03:08:44 comma-1a8caafe unknown: [hardware] engaged: False                                                                                                                
Jan 22 03:08:44 comma-1a8caafe unknown: [hardware] engaged: False                                                                                                                
Jan 22 03:08:45 comma-1a8caafe unknown: [hardware] engaged: False                                                                                                                
Jan 22 03:08:45 comma-1a8caafe systemd[1]: Reloading finished in 1442 ms.                                                                                                        
Jan 22 03:08:45 comma-1a8caafe NetworkManager[4311]: <info>  [1737515325.8187] audit: op="connection-add" uuid="5702b0b8-14ef-4825-bda1-a6ff56bc3823" name="Hotspot" pid=50749 ui
d=1000 result="success"
```

```
comma@comma-1a8caafe:/data/openpilot$ time nmcli c delete Hotspot
Connection 'Hotspot' (75aaff2c-3659-45a9-84e8-078070511473) successfully deleted.
Connection 'Hotspot' (312daf32-8ecf-4f7e-94ae-d080e81f78ac) successfully deleted.
Connection 'Hotspot' (d1887a25-4662-493c-947d-fbbb7f35e99b) successfully deleted.
Error: Timeout expired (10 seconds)
```

adeebshihadeh, 2025-01-23:
Seems like NM actually just got slower...
```
batman@workstation-adeeb:~$ time nmcli connection add type ethernet ifname eth1 con-name tmp && time nmcli con delete tmp
Connection 'tmp' (944dd592-0632-470b-8976-4709c05f0f22) successfully added.

real	0m0.668s
user	0m0.007s
sys	0m0.008s

Connection 'tmp' (944dd592-0632-470b-8976-4709c05f0f22) successfully deleted.

real	0m0.659s
user	0m0.008s
sys	0m0.007s
```

closed 2025-01-23 by adeebshihadeh