commaai/openpilot #34295: test_onroad stress enhancemnets

adeebshihadeh, 2024-12-20:
Now that test_onroad is reliable, it's time to add some things to really stress the system:
* lots of disk load: openpilot shouldn't rely on a responsive disk. catches things like reading params in a real-time loop
* change the system time: catch things like https://github.com/tinygrad/tinygrad/commit/cda34ccadf07181422dfb9ff38470d1454cfff1e

JacksonRT, 2025-01-19:
@adeebshihadeh Would writing large files to the disk for the test duration the removing the afterward be what your looking for?

JacksonRT, 2025-01-19:
@adeebshihadeh This is what I've got so far.

  def create_disk_load(file_path="temp_file.txt", file_size_mb=50000):

    file_size_bytes = file_size_mb * 1024 * 1024
    data = os.urandom(file_size_bytes)
    #Create File and Writes/Reads its contents
    try:
      with open(file_path, "wb") as f:
        f.write(data)
      with open(file_path, "rb") as f:
        f.read()
    #Remove File afterwards
    finally:
      try:
        os.remove(file_path)
      except FileNotFoundError:
        # Ignore if the file doesn't exist
        pass

  if __name__ == "__main__":
    create_disk_load()