commaai/openpilot #34300: Uninstall button in addition to Reboot in Failed to Build screen

nelsonjchen, 2024-12-21:
https://github.com/commaai/openpilot/issues/27041

[image]

The `Reboot` button in this screen more or less never really solves the issue ever. Uninstall at least lets the user wipe and refresh their devices easily from either a working branch or maybe the branch that broke broke the updater and reinstalling it is the fastest way to get working.

Users can get into this state slightly periodically. This can also happen on master-ci but it can also very much happen in topical branches as OP. 

The alternative is telling users to perform a device reset by tapping the at boot to ask for a reset. While that may work, it's not very obvious and takes some explanation to do which in cases of mass Failed to Build, is tiresome to explain on Discord. It's also a bit physical sometimes. I think I might have even contributed to the wearing out of my first mount too much from doing this and I'm a permanent install user. 

But mostly, this issue exists because every time I see that screen, I can't help but look at that `Reboot` button and think you are one of the most useless buttons ever. There could be a much better button there or adjacent to it for Uninstall. It feels like it should be there.

nelsonjchen, 2024-12-21:
Alternatively, it'll be neat too if instead of uninstall, it had a fetch and force reset button. That might be a separate issue or maybe this issue could be changed to that.

adeebshihadeh, 2024-12-21:
You really just want this: https://github.com/commaai/agnos-builder/issues/156

closed 2024-12-21 by adeebshihadeh

nelsonjchen, 2024-12-22:
Well, then this issue has a good suggestion for the first sentence of that issue. On this screen should be one of the entry points to that AGNOS settings menu at least.