commaai/openpilot #34318: comma 3x br

josooik, 2024-12-28:
### Describe the bug

Hello.

Black screen during comma 3x open pilot installation.

Attempt to initialize https://flash.comma.ai/ failed.

I would appreciate it if you could tell me how to solve it.
[image]
[image]

### Provide a route where the issue occurs

3fe4b2f9c784287e

### openpilot version

0.9.7

incognitojam, 2025-01-04:
What steps have you tried so far? Have you tried different cable, USB port or computer as suggested?

josooik, 2025-01-05:
Comma 3x Brick State (Black Screen) Flash Comma does not work.

The solution is

git clone https://github.com/commaai/agnos-builder.git
cd agnos-builder
git submodule update --init agnos-kernel-sdm845
./tools/extract_tools.sh
./build_kernel.sh
./flash_kernel.sh(comma 3/3x QDL MODE 진입)
If you succeed up to six times, you will be able to enter Fastboot with a comma log when you give OBD-C port power five minutes later.
You can flash it at https://flash.comma.ai/ and recover it.

closed 2025-01-05 by josooik