commaai/openpilot #34322: Pay attention alert with passengers

jschaffer87, 2024-12-31:
Any guidance on false "pay attention" alerts with a passenger. With someone in passenger seat, they can't look away from the road, like the driver and the C3 screams constantly. The only way to get it to not acknowledge the passenger is leaning all the way forward towards the dash. The software should have a selection for right or left hand drive.

eggehad, 2025-01-01:
I installed my comma 3x today for the first time. As long as the driver is the only one in the car. It works great. However this evening my wife was in the front passenger seat, playing with her phone. She was not driving.  Comma screamed and shutdown, refusing to work, due to lack of attention. This is a disaster, that is the exact scenario I need comma to work in.  Trips. With a car full of people!   I can't believe this problem hasn't come up before!

nelsonjchen, 2025-01-02:
https://github.com/commaai/openpilot/wiki/FAQ#how-do-i-report-a-bug

Make sure to include data.

In this report you will find someone reported an issue but it hadn't been replied to and they found instructions to force the side https://discord.com/channels/469524606043160576/1296873752780607579/1296873752780607579 . idk, it's a bit in the grey area of adjusting dm to force a side which is somewhat verboten.

that said, you should still #submit-feedback so at least someone from comma could look at it and fix the model.

adeebshihadeh, 2025-01-20:
The model in master is targeted at fixing this; you can try it early with the nightly build or you can wait for release in a couple weeks.