commaai/openpilot #34331: Get rid of openpilot.common.numpy_fast

adeebshihadeh, 2025-01-06:
This optimization likely no longer makes sense in the post comma two world. After it's done, we should do some casual profiling to make sure we really don't need it.

It should also be removed from [opendbc](https://github.com/commaai/opendbc/blob/master/opendbc/car/common/numpy_fast.py).

adeebshihadeh, 2025-01-06:
I see in https://github.com/commaai/openpilot/pull/34332 that there's tons of casts from numpy to Python types. Can we just patch `pycapnp` to handle this for us?

Shaikimram, 2025-01-10:
hey @adeebshihadeh, I would like to take up this issue.

sunnyhaibin, 2025-01-10:
> hey @adeebshihadeh, I would like to take up this issue.

Open a PR. You do not need to request for permission to work on the issue.

Shaikimram, 2025-01-10:
> > hey [@adeebshihadeh](https://github.com/adeebshihadeh), I would like to take up this issue.
> 
> Open a PR. You do not need to request for permission to work on the issue.

alright

deanlee, 2025-01-12:
> I see in #34332 that there's tons of casts from numpy to Python types. Can we just patch `pycapnp` to handle this for us?

This approach can introduce performance overhead, as type checking occur with every assignment.  and may slow down the overall process.

Sammohana, 2025-01-12:
> 

I opened a [PR](https://github.com/capnproto/pycapnp/pull/373) in [pycapnp](https://github.com/capnproto/pycapnp) repo for pycapnp to handle numpy data type.

Shaikimram, 2025-01-12:
hey @Sammohana , can you share your discord ID, I want to connect with you for some time

Sammohana, 2025-01-12:
> hey @Sammohana , can you share your discord ID, I want to connect with you for some time

brainlesspea

deanlee, 2025-01-13:
> > I see in #34332 that there's tons of casts from numpy to Python types. Can we just patch `pycapnp` to handle this for us?
> 
> I opened a [PR](https://github.com/capnproto/pycapnp/pull/373) in [pycapnp](https://github.com/capnproto/pycapnp) repo for pycapnp to handle numpy data type conversion. This is likely going to take few months should introduce another lib to handle this or use my [repo](https://github.com/sammohana/pycapnp)

I feel that the PR you submitted to pycapnp is a bit too complex, which might make it difficult to merge.

Sammohana, 2025-01-14:
> I feel that the PR you submitted to pycapnp is a bit too complex, which might make it difficult to merge.

Okay I will work on this

closed 2025-01-14 by adeebshihadeh