commaai/openpilot #34346: 22 Acura RDX - steering disables under 45mph

mvl-boston, 2025-01-09:
### Describe the bug

Steering is completely disabled below 45mph on release branch.   There is no visual warning saying it is disabled, the car just drives in a straight line until it about to go offroad and reports a manual steering required warning.

I have traced the RCA to the release branch requiring factory ACC for longitudinal control.   The factory ACC disables LKAS below 45mph, and therefore it appears the car ignores all comma steering commands.    If install the latest devel branch, and enable comma longitudinal control instead of factory, then steering works at all speeds.

I would expect some type of warning under 45mph, rather than a green path that the car just ignores.   Or an preferred fix is to default release branch to comma longitudinal for this car model, either always or whenever under 45mph.

Hardware used was a retail comma 3x with included bosch A harness, and not using included comma power.

### Which car does this affect?

Acura RDX 2022

### Provide a route where the issue occurs

all routes / all locations

### openpilot version

0.9.7

### Additional info

The 2022 RDX is listed in the official compatibility chart as compatible with a mention of a 3mph steering limitation.   In reality the limitation is under 45mph for anyone on the release branch with no override available, so a purchaser would not be getting the experience they expect.

I'm also surprised that the supported models listed on comma's compatibility chart reference the 2019-2022.   Acura redesigned the RDX for 2022 and that redesign remained unchanged until 2024.   I suspect this issue will occur in all 2022-2024 RDX models, and have a similar workaround.   It may not occur from 2019-2021 as that is likely a different ACC/LKAS implementation.

There appears to be a similar issue reported for the 5th gen Honda Odyssey below.
https://discord.com/channels/469524606043160576/1260687525744476223/1260693383379157025

mvl-boston, 2025-01-10:
Upon more searching, I see much of this was planned to be addressed in a pull request below, but it was never rewritten for opendbc.

https://github.com/commaai/openpilot/pull/25481

I'm trying to dig into the guts of the prior proposal to see if I can follow it well enough to write a proposed opendbc variant.  Until a similar change is made/pulled, I'd recommend comma revise it's official published compatibility list to remove the 2022 RDX as this can be a safety issue if a driver is misled into thinking that steering is occurring.

mvl-boston, 2025-01-10:
Route added

jyoung8607, 2025-01-10:
> Route added

Thank you, and I see it's marked public. Can you go ahead and upload all logs for this route? Without rlogs, we don't have the vehicle CAN traffic to look at, and we can't run the route through offline replay tools to investigate and fix the issue. In comma connect, Files->Upload all logs.

mvl-boston, 2025-01-13:
I uploaded this morning.   Let me know if you need anything further.

jyoung8607, 2025-01-14:
We'll track this under commaai/opendbc#1637.

closed 2025-01-14 by jyoung8607