commaai/openpilot #34348: 22 Acura RDX - max speed on comma mismatches max speed in instrument cluster - possibly mph/kph error

mvl-boston, 2025-01-09:
### Describe the bug

The 2022 Acura RDX shows the max speed on the instrument cluster, plus comma shows the max speed on the screen, but these max speeds do not match.

The instrument cluster is a little under double the comma screen speed.   Possibly by the mph vs kph conversion factor.   eg: When the instrument cluster shows a max mph of 60, the comma shows a max mph of about 36.

Using the speed up/down toggle on the steering wheel changes both numbers, and they maintain their mismatch ratio.

When actually driving, the release branch (which uses factory ACC for longitudinal) maxes its driving at the instrument cluster's physical speed.   I install devel branch and toggle override to comma longitudinal control, then the max driving is the comma screen/lower speed.

Expected result is that the max speed on the car instrument cluster matches the max speed on the comma screen.

Hardware is a retail comma 3x with included bosch A harness, and not using included comma power.

### Which car does this affect?

Acura RDX 2022

### Provide a route where the issue occurs

all routes / all locations

### openpilot version

0.9.7 - both release and devel

mvl-boston, 2025-01-10:
Looks like this is the same issue in the bug below:
https://github.com/commaai/openpilot/pull/25450

I wrote a terrible hack in my mvl-boston/devel branch that proves this is the root cause, and works around the issue.   Now the harder part is to segregate the 22 RDX fingerprint from the 19-21 fingerprint so this fix can be properly written.

Hopefully someone more familiar with the fingerprinting can do this.

jyoung8607, 2025-01-10:
> all routes / all locations

This isn't optional. We can't begin to look into the issue without a recorded route that demonstrates the problem.

https://github.com/commaai/openpilot/wiki/FAQ#how-do-i-report-a-bug

Make sure to upload all logs for the route via comma connect. If you mark the route as public, this will greatly expand the number of people that can help.

mvl-boston, 2025-01-10:
Route added.

jyoung8607, 2025-01-10:
> Route added.

This route doesn't have rlogs uploaded, but I see you're using the same route in #34346, so as soon as the logs are uploaded we can look at both issues.

mvl-boston, 2025-01-12:
I tried uploading and they were pending until the comma turned back on (as noted I don't use comma power).  I ran comma for a few drives today, and it didn't seem to upload.  (unless you tell me you can see them on your end)

Do logs need to be uploaded from release software only?   The original issue was recorded on the official 0.9.7 release branch, however the device is currently running my mvl-boston/devel branch, which is a fork of the 0.9.7 official devel branch with one line / one character changed to flip max kph to mph.   Can a devel branch upload logs?

jyoung8607, 2025-01-12:
Problem reports from release, devel, or master/master-ci are all useful and actionable, and you can always upload logs. That said, the route in question is from nine days ago, and the space may have been reclaimed if you've driven for several hours since then, so that may be why the logs aren't coming up. Just try uploading logs from a more recent route.

mvl-boston, 2025-01-13:
I did another upload and it said 27 logs sucessfully uploaded.  I had the route preserved.

Can you see if ypu got what you needed here?

jyoung8607, 2025-01-13:
I'm afraid not. Your device only retains a few hours worth of driving. Please try uploading logs from a more recent route.

If in doubt, look at More Info->View in useradmin. Your route has qlogs only:

[image]

We need the rlogs in order to troubleshoot. Here's an example from one of my devices:

[image]

mvl-boston, 2025-01-13:
Thanks, all my recent drives were on the mvl-boston/devel fork, which is openpilot's 0.9.7 devel build with one character change (mph to kph) in 1 line of code to workaround the bug.

I see the notes on top that no forks are allowed, is this close enough so you can use a recent drive?   Or do I need to uninstall/reinstall a true openpilot build to create a new route for debugging?

jyoung8607, 2025-01-13:
That's fine. It's mainly a problem with forks that mess with capnp/cereal structures that break the tools we use to parse logs. I'll take what you have.

mvl-boston, 2025-01-14:
Requested upload of a more recent route and revised the route ID above.   Should show up by tomorrow at the latest when we power on and drive the car.

For my other bug I'll need to drive tomorrow with longitudinal off to get you a segment to use.

jyoung8607, 2025-01-14:
We'll track this under commaai/opendbc#1637.

closed 2025-01-14 by jyoung8607