commaai/openpilot #34365: stuck in "registering device" when downgrading from master to release

jyoung8607, 2025-01-12:
### Describe the bug

**Steps to reproduce**

1. Start with current master-ci on AGNOS 11.4, on a device without comma prime/LTE connectivity
2. Factory reset, or uninstall from the UI
3. Attempt a fresh install of release openpilot (no custom software), as of this writing openpilot 0.9.7
4. openpilot startup spins in "registering device".

**Workaround**

Another factory reset will allow install and registration to finish.

### Provide a route where the issue occurs

N/A

### openpilot version

08779941e533ce4cb36410c899b3dff3883d59fe (today's master)

### Additional info

The proximate cause is commaai/agnos-builder#423, but it's also not clear that we should spin forever waiting on registration that isn't happening. I'm not familiar enough with the process to know if skipping it could be harmful.

adeebshihadeh, 2025-01-12:
So registration is [effectively dead](https://github.com/commaai/openpilot/blob/55cdf5ad7c297ffe818ed045fa99b7a58cfb6eb7/system/athena/registration.py#L25) for most comma 3X, but early comma 3X and all comma three need it until the necessary backend changes are implemented to entirely remove it.

The looping is a problem that's always been there, and I'd like to solve it by killing registration. Even if we don't get around to the backend changes, I'm not too worried about it.

jyoung8607, 2025-01-12:
It looks like a [reasonably small tweak](https://github.com/commaai/agnos-builder/issues/423#issuecomment-2585865488) to AGNOS 11 will fix backward compatibility and sidestep the registration hang.

adeebshihadeh, 2025-01-12:
Closing since we don't plan on doing anything about the looping. We'll track this in the agnos-builder issue.

closed 2025-01-12 by adeebshihadeh