commaai/openpilot #34377: CAN BUS disconnected likely faulty cable due to CAM sync error?

porteken, 2025-01-13:
### Describe the bug

I keep getting a “CAN BUS disconnected likely faulty cable” error.  Looking at the route log, I see these two errors multiple times: "CAM_SYNC error: id 6 - errno 11 - ret 0 - ioctl_result -110"  and "failed to wait for sync: -110 3"

### Provide a route where the issue occurs

a5eab6e9192ef61d/00000043--7d54bb5234/0

### openpilot version

0.9.8

jyoung8607, 2025-01-13:
In comma connect, please mark the route so that others can see it: More info->Public access

Also, if not already done, make sure that all logs are uploaded with the device on Wi-Fi: Files->All logs->Upload

porteken, 2025-01-13:
I have made the route public and uploaded all of the logs.

jyoung8607, 2025-01-14:
This isn't an officially supported car, but we hope to change that soon and there's a bunch of them known to be driving okay on @vanillagorillaa's fork, so for now we'll proceed on the assumption it should work.

There's definitely some sort of physical issue with CAN connectivity. CAN bus 0 (on transceiver 2, flipped OBD-C orientation) sees intermittent bursts of errors followed by complete loss of connectivity later in the route.

Per the error message, I trust you've tried replacing the cable with another suitable USB 3.1 gen 2 cable? If not, that's your next troubleshooting step. That said, between this and the CAM sync errors, I'm a little suspicious of the device.

I'm going to mark this Closed, not because of the unsupported car, but because all signs point to a cable or device issue rather than a software bug in openpilot. Once you've verified the problem keeps happening with another known-good cable, reach out to comma.ai/support for warranty or repair options.

[image]

closed 2025-01-14 by jyoung8607

porteken, 2025-01-14:
Where did you get the graph?

jyoung8607, 2025-01-14:
1. Set up an [openpilot development environment](https://github.com/commaai/openpilot/blob/master/tools/README.md) on your local machine
2. Start [PlotJuggler (from openpilot tools)](https://github.com/commaai/openpilot/tree/master/tools/plotjuggler) on the route in question
3. Load the CAN bus debug layout from [saved layouts](https://github.com/commaai/openpilot/blob/master/tools/README.md) to visualize the data