commaai/openpilot #34392: Text input dialog for `/data/params/d/SecOCKey`

calvinpark, 2025-01-15:
Owners of Toyota Security Key (TSK) vehicles need to extract and write the security key to `SecOCKey` param.

The biggest hurdle for this is the SSH requirement, as many people have not used SSH before.

There's a community effort to extract the key using a GUI application, to remove the SSH requirement for the key extraction.

Even after SSH is set up and the key is extracted, reinstalling remains to be cumbersome. Uninstalling openpilot also deletes the key file, so the users must SSH in and write the key again on each uninstall/reinstall.

Please make a settings option that allows users to type in their security key to be saved in `SecOCKey` param. This will allow novice users to avoid using SSH entirely.

Here's a POC (https://github.com/sunnypilot/sunnypilot/pull/575) that I'd be happy to change to comma's requirements and submit as a PR.

Thank you

Chaim8, 2025-01-15:
This tool is designed to help those who are new to SSH, making the experience easier and more accessible. Our hope is that one day, we can turn the whole process of key extraction and dumping into a simple one-click solution. We're excited for you to get started!

jyoung8607, 2025-01-16:
> There's a community effort to extract the key using a GUI application, to remove the SSH requirement for the key extraction.
>
> Even after SSH is set up and the key is extracted, reinstalling remains [...]

On the one hand, modifying the UI to collect and store a text string would be pretty lightweight. On the other hand, entering a 16 byte hexadecimal string by hand is going to be far from user friendly.

If there's going to be an external application that solves extraction, storage, and installation, why not just use that same application for reinstallation when necessary?

adeebshihadeh, 2025-01-16:
Gonna close since this isn't a bug. We're supportive of making this experience better, but it should be in the form of a PR (we're not merging a SecOc keyboard).

closed 2025-01-16 by adeebshihadeh

calvinpark, 2025-01-16:
@jyoung8607 There is an external application, TSK Manager, but the only way to run TSKM is by first uninstalling OP and then installing TSKM. The keyboard makes it such that the user does not need to uninstall OP to enter the key.

SunnyPilot is preparing for a feature where they back up the file content to their own cloud SunnyLink and then restore automatically. That too would solve the problem, but only for SP. I was hoping that SecOC Keyboard would make such a mechanism unnecessary.