commaai/openpilot #34409: ui: preview driver monitor crashes UI on exit

sunnyhaibin, 2025-01-17:
### Describe the bug

- Possibly introduced in https://github.com/commaai/openpilot/pull/33375

https://github.com/user-attachments/assets/3a929493-34ff-46ed-a3b8-2841d2b01155

### Provide a route where the issue occurs

69cd3a350561a4d8

### openpilot version

c5150215762099526e8d518f8ce53923d59ed812

deanlee, 2025-01-18:
I believe this issue is not related to the UI itself. The UI is being terminated by the watchdog due to a VIPCissue, which may occur after repeated connect/disconnect with the VIPC server. https://github.com/commaai/msgq/pull/624

deanlee, 2025-01-18:
There are actually several different issues causing the UI to crash and trigger a restart:

1. https://github.com/commaai/openpilot/issues/34411
2. vipc issues: https://github.com/commaai/msgq/pull/624. which causes an assertion failure on repeated connect/disconnect:
`  ui: msgq_repo/msgq/visionipc/visionipc_client.cc:128: static std::set<VisionStreamType> VisionIpcClient::getAvailableStreams(const std::string &, bool): Assertion `r == sizeof(request)' failed.`