commaai/openpilot #34449: ci: Jenkins is flaky again

maxime-desroches, 2025-01-22:
`test_onroad` issues: 
- Higher memory usage (from the `tinygrad` merge) ([run](https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/master/185/pipeline))
- Out of bounds qlog size
- `modelV2` exec time ([run](https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/master/214/pipeline#step-338-log-968))
- `ubloxd` crashing ([run](https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/master/209/pipeline/))
- `camerad` crashing ([run](https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/master/195/pipeline))

`test_panda_spi` issues:
- Get `RuntimeWarning: Mean of empty slice` from numpy in a print statement ([run](https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/master/206/pipeline#step-426-log-155))

`test_power_draw` issues:
- `camerad` sending 0 messages ([run](https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/master/154/pipeline#step-580-log-1099))

adeebshihadeh, 2025-02-17:
Did a fresh run and only two failures now:
- [ ] 1x modelV2 timings [here](https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/__jenkins_loop_master/31/pipeline/)
- [ ] 20x OX03C10 `test_camerad` failures [here](https://jenkins.comma.life/blue/organizations/jenkins/openpilot/detail/__jenkins_loop_master/26/pipeline/476)

adeebshihadeh, 2025-02-21:
All good now after the last bit of camerad work

closed 2025-02-21 by adeebshihadeh

maxime-desroches, 2025-02-21:
`test_onroad` now takes 56-57s which is very close to the 1 minute timeout. It used to take 48s very consistently.

adeebshihadeh, 2025-02-21:
Oh I added some more tests, is that why? There should be no overhead