commaai/openpilot #34475: Experimental Mode | Reset Settings | Every Restart

LANdrew12, 2025-01-25:
### Describe the bug

The last 3 versions of OP (ever since the button for long was moved to developer menu), every time the car restarts it unchecks that option from the menu putting it back into chill mode. 
The version I am on: 0.9.8 / master / dcb05af / Jan 22

### Provide a route where the issue occurs

a4653a9be878a408/00000017--c16255a167/0

### openpilot version

0.9.8

### Additional info

The route isnt really related. Its every drive, every time.

LANdrew12, 2025-01-25:
I have a volkswagen jetta 2020 sel premium
Let me know if more info is needed

BBBmau, 2025-01-25:
this was addressed and fixed earlier today: https://github.com/commaai/openpilot/pull/34467

closed 2025-01-25 by adeebshihadeh