commaai/openpilot #34483: openpilot Unavailable Waiting to start

aRossAllen, 2025-01-26:
### Describe the bug

R4P 2021 Toyota Comma 3x
After downloading ECU keys
Writning ECU Key to SecOCkey
Briefly working (3 weeks)
Then after a refflash
Install master-ci again would not run
Have tried sunny pilot, Master-ci but give same error.
Submitted ticket to Comma they took a look at my logs and said not a hardware issue with a screenshot with no expiation.
Suggested community support for this ?

### Provide a route where the issue occurs

https://useradmin.comma.ai/?onebox=762a536115146c5d%7C00000001--f59a8f131c

### openpilot version

0.9.8

### Additional info

currently on SunnyPilot but thinking of trying FrogPilot

adeebshihadeh, 2025-01-26:
Closing as this isn't a bug - check out our [Discord](https://discord.comma.ai) instead.

closed 2025-01-26 by adeebshihadeh

jyoung8607, 2025-01-26:
Something went wrong saving your key, openpilot can't parse it. If in doubt about the process, ask in #toyota-security, there have been some recent changes.

@sshane can you transfer this issue to opendbc? The user who opened the issue can fix , but `card` should handle the ValueError exception without faceplanting. 

```
{'errorLogMessage': '{"msg": "crash", "ctx": {"daemon": "card", "dongle_id": "762a536115146c5d", "version": "0.9.8", "origin": '
                    '"github.com/commaai/openpilot", "branch": "master-ci", "commit": "e083f2da795d0631cdbd6584fd548199d0aec02f", "dirty": false, '
                    '"device": "tizi"}, "exc_info": "Traceback (most recent call last):\\n  File '
                    '\\"/data/openpilot/openpilot/system/manager/process.py\\", line 41, in launcher\\n    mod.main()\\n  File '
                    '\\"/data/openpilot/selfdrive/car/card.py\\", line 274, in main\\n    car = Car()\\n          ^^^^^\\n  File '
                    '\\"/data/openpilot/selfdrive/car/card.py\\", line 129, in __init__\\n    saved_secoc_key = '
                    'bytes.fromhex(secoc_key.strip())\\n                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\\nValueError: non-hexadecimal number '
                    'found in fromhex() arg at position 0", "level": "ERROR", "levelnum": 40, "name": "swaglog", "filename": "process.py", "lineno": '
                    '47, "pathname": "/data/openpilot/openpilot/system/manager/process.py", "module": "process", "funcName": "launcher", "host": '
                    '"comma-8b8c4589", "process": 48981, "thread": 548488196128, "threadName": "MainThread", "created": 1735653149.0213544}',
```

aRossAllen, 2025-01-26:
Great - Thankyou for your help!!