commaai/openpilot #34492: Problem with H.264 bitstream malformed, no startcode found

panminjia82, 2025-01-29:
### Describe the bug

RUN OpenPilot's master branch on PC with two webcam@30fps on mjpg model.

error log 

000019- - 6350628e80--0
Getting VIN & FW versions
webcamerad logcatd proclogd logmessaged mic dmonitoringmodeld encoderd loggerd modeld ui soundd locationd pandad calibrationd torqued controlsd selfdrived card deleter monitoringd paramsd plannerd radard hardwared statsd
Setting OBD multiplexing to True Starting listener for: camerad
system/loggerd/encoderd.cc: encoder road_cam_encoder init 640x480
OBD multiplexing set successfully
[mpegts @ 0x62c5330c6c00] H.264 bitstream malformed, no startcode found, use the video bitstream filter 'h264_mp4toannexb' to fix it ('-bsf:v h264_mp4toannexb'
option with ffmpeg)
system/loggerd/video_writer.cc: ts encoder write issue len: 67624 ts: 50000
[mpegts @ 0x62c5330c6c00] H.264 bitstream malformed, no startcode faund, use the video bitstream filter 'h264_mp4toannexb' to fix it ('-bsf:v h264_mp4toannexb'
option with ffmpeg)
system/loggerd/video_writer,cc: ts encoder write issue len: 67616 ts: 100000
[mpegts @ 0x62c5330c6c00] H.264 bitstream malformed, no startcode found, use the video bitstream filter 'h264_mp4toannexb' to fix it ('-bsf:v h264_mp4toannexb'

### OS Version

Ubuntu 24.04

panminjia82, 2025-02-01:
@deanlee
Thank you for your help .

After commit your code ,Warning of"non-strictly-monotonic PTS" still occurred .and the WedCam DO NOT WORK
here is the log :
(openpilot) panminjia@Lenovo-M75n:/data/openpilot$ NOSENSOR=1 USE_WEBCAM=1 system/manager/manager.py
system/loggerd/encoder/ffmpeg_encoder.cc: needs update
system/loggerd/bootlog.cc: bootlog to /home/panminjia/.comma/media/0/realdata/boot/000000b3--cdf2305b68
missing public key: /home/panminjia/.comma/persist/comma/id_rsa.pub
: Could not find the Qt platform plugin "wayland" in ""
Panda 3b004e001751373134373734 connected, version: DEV-3ff97305-DEBU, signature 0251e6f044051fc7, expected 0251e6f044051fc7
selfdrive/pandad/main.cc: starting pandad
selfdrive/pandad/pandad.cc: connecting to pandas: 3b004e001751373134373734
selfdrive/pandad/panda.cc: connected to 3b004e001751373134373734 over USB
selfdrive/pandad/pandad.cc: connected to all pandas
logmessaged ui deleter pandad hardwared statsd
opening roadCameraState at /dev/video0
Waiting for CAN messages...
system/loggerd/loggerd.cc: logging to /home/panminjia/.comma/media/0/realdata/00000058--af50d1b259--0
webcamerad logcatd proclogd logmessaged micd dmonitoringmodeld encoderd loggerd modeld ui soundd locationd calibrationd torqued controlsd selfdrived card deleter dmonitoringd pandad paramsd plannerd radard hardwared statsd
Getting VIN & FW versions
Setting OBD multiplexing to True
OBD multiplexing set successfully
webcamerad logcatd proclogd logmessaged micd dmonitoringmodeld encoderd loggerd modeld ui soundd locationd calibrationd torqued controlsd selfdrived card deleter dmonitoringd pandad paramsd plannerd radard hardwared statsd
got vin with request=b'\t\x02'
opening driverCameraState at /dev/video2
Setting OBD multiplexing to False
Starting listener for: camerad
system/loggerd/encoderd.cc: encoder driver_cam_encoder init 640x480
system/loggerd/encoderd.cc: encoder road_cam_encoder init 640x480
[libx264 @ 0x75f200005a80] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2 AVX FMA3 BMI2 AVX2
[libx264 @ 0x75f200005a80] profile High, level 2.1, 4:2:0, 8-bit
OBD multiplexing set successfully
modeld init
setting up CL context
CL context ready; loading model
webcamerad logcatd proclogd logmessaged micd dmonitoringmodeld encoderd loggerd modeld ui soundd locationd calibrationd torqued controlsd selfdrived card deleter dmonitoringd pandad paramsd plannerd radard hardwared statsd
models loaded, dmonitoringmodeld starting
connecting to driver stream
connected with buffer size: 460800
iso-tp query bad response: (1793, None) - 0x7f3e13
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
iso-tp query bad response: (1792, None) - 0x7f3e13
iso-tp query bad response: (1793, None) - 0x7f3e13
iso-tp query bad response: (1988, None) - 0x7f0911
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
iso-tp query bad response: (1920, None) - 0x7f0910
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
iso-tp query bad response: (1988, None) - 0x7f3e11
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
iso-tp query bad response: (1968, None) - 0x7f3e12
iso-tp query bad response: (1953, None) - 0x7f3e12
iso-tp query bad response: (1920, None) - 0x7f3e12
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
iso-tp query bad response: (1872, 109) - 0x7f3e12
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
webcamerad logcatd proclogd logmessaged micd dmonitoringmodeld encoderd loggerd modeld ui soundd locationd calibrationd torqued controlsd selfdrived card deleter dmonitoringd pandad paramsd plannerd radard hardwared statsd
iso-tp query bad response: (1872, 15) - 0x7f3e12
VIN LVGB674K4KG018519
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[... 34 lines cut ...]
skipping model eval. Dropped 2 frames
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
skipping model eval. Dropped 5 frames
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS
webcamerad logcatd proclogd logmessaged micd dmonitoringmodeld encoderd loggerd modeld ui soundd locationd calibrationd torqued controlsd selfdrived card deleter dmonitoringd pandad paramsd plannerd radard hardwared statsd
skipping model eval. Dropped 1 frames
skipping model eval. Dropped 4 frames
webcamerad logcatd proclogd logmessaged micd dmonitoringmodeld encoderd loggerd modeld ui soundd locationd calibrationd torqued controlsd selfdrived card deleter dmonitoringd pandad paramsd plannerd radard hardwared statsd
skipping model eval. Dropped 3 frames
skipping model eval. Dropped 2 frames
[... cut ...]
webcamerad logcatd proclogd logmessaged micd dmonitoringmodeld encoderd loggerd modeld ui soundd locationd calibrationd torqued controlsd selfdrived card deleter dmonitoringd pandad paramsd plannerd radard hardwared statsd
skipping model eval. Dropped 4 frames
webcamerad logcatd proclogd logmessaged micd dmonitoringmodeld encoderd loggerd modeld ui soundd locationd calibrationd torqued controlsd selfdrived card deleter dmonitoringd pandad paramsd plannerd radard hardwared statsd
skipping model eval. Dropped 3 frames
skipping model eval. Dropped 1 frames
skipping model eval. Dropped 3 frames

deanlee, 2025-02-01:
```
H.264 bitstream malformed, no startcode found, use the video bitstream filter 'h264_mp4toannexb' to fix it ('-bsf:v h264_mp4toannexb'
option with ffmpeg)
```
This is an encoder issue, not related to the webcam. If you're facing webcam issues, make sure the correct camera IDs are set properly.

The `[libx264 @ 0x75f200005a80] non-strictly-monotonic PTS `warning can be safely ignored for now. Despite the warning, H.264 video (qcamera) will still be logged after https://github.com/commaai/openpilot/pull/34498. This is a separate issue and will be addressed in a future PR.

panminjia82, 2025-02-01:
Thank you very much.I will  ignore it .

webcam IDs maybe is not the reason.
Driver_cam and Road_cam  work normally.

the ui message is "Sensor Data Invalid
Possible Hardware Issue".

I do  not have any IMU sensor ,this  occur issue ?
OR,  frames are droppedthat make the issue?

deanlee, 2025-02-01:
> the ui message is "Sensor Data Invalid Possible Hardware Issue".
> 
> I do not have any IMU sensor ,this occur issue ?

Yes, the UI message "Sensor Data Invalid" appears because the system can't detect accelerometer or gyroscope sensors. Since you're running this on a PC, which doesn't have these sensors, this message is expected and not indicative of an actual hardware issue.

closed 2025-02-11 by incognitojam