commaai/openpilot #34536: Try reverting "compress on upload"

adeebshihadeh, 2025-02-06:
https://github.com/commaai/openpilot/pull/34528 broke compressed logs. It wasn't caught in CI since we use what's on disk in CI, which isn't what gets uploaded. We can add tests, but I think we should also investigate moving the compression back to loggerd.

IIRC we did that for two reason:
1. CPU usage
2. Getting as much as possible in the case of sudden power loss

#1 should be much better with zstd, and #2 is better defined now.

adeebshihadeh, 2025-02-06:
@deanlee want to try this out?

closed 2025-02-08 by adeebshihadeh

deanlee, 2025-02-10:
> [#34528](https://github.com/commaai/openpilot/pull/34528) broke compressed logs

This is because the original LogReader could not decompress streaming zstd compressed logs. This issue has been resolved in [PR #34549](https://github.com/commaai/openpilot/pull/34549).