commaai/openpilot #34599: Replay: Bus Error(core dumped) when replay demo

feihero1003, 2025-02-17:
### Describe the bug

I tried V0.9.7 on PC (docker + ubuntu20.04), but error "bus error(core dumped)" was raised then replay was terminated.
Linux version:
[image]

'replay' execution:
[image]

gdb info:
[image]

### OS Version

Ubuntu 20.04

### openpilot version or commit

50dd024081496cc5f036a99c8f62d2887112dbb7

deanlee, 2025-02-17:
This looks like an OOM ssue, especially since you're using Docker. Check your container and system memory usage to ensure there's enough available during the replay.

feihero1003, 2025-02-17:
seems the memory status of container and host is fine:

[image]

docker stats result:
[image]
[image]

feihero1003, 2025-02-17:
> This looks like an OOM ssue, especially since you're using Docker. Check your container and system memory usage to ensure there's enough available during the replay.

thanks so much, after config --shm-size, it works now.

closed 2025-02-17 by feihero1003