commaai/openpilot #34648: camerad: SOF/EOF got misaligned

adeebshihadeh, 2025-02-21:
At the start of the route, the frame ID skips four frames, however the SOF and EOF get out of sync and seem to be for different frames.

SOF and EOF get handled in two different places, so that's somewhere to start.

We may have happened to fix this with the recent camerad refactors, but we should root cause nonetheless. Attached is the rlog from the route.

[8011d605be1cbb77_00000056--42d16b4e5c--0--rlog.zip](https://github.com/user-attachments/files/18900455/8011d605be1cbb77_00000056--42d16b4e5c--0--rlog.zip)

adeebshihadeh, 2025-02-21:
@deanlee want to check this out?

deanlee, 2025-02-21:
I believe the misalignment issue was caused by two factors:
1. Frame metadata was being read and written in different threads. The main thread could overwrite the metadata while the camera thread was popping the index from the queue and reading it, leading to potential data races. This issue has been resolved by https://github.com/commaai/openpilot/pull/34627
2. The assignment of EOF was not properly validated to ensure that EOF actually needed updating. The frame metadata was being overwritten unconditionally during `enqueue_req_multi` calls, but it should only be updated if the index needed to be pushed to the queue.

closed 2025-02-21 by adeebshihadeh