commaai/openpilot #34663: camerad: BPS config failure

adeebshihadeh, 2025-02-22:
Just happened on master

https://commaai.sentry.io/issues/6012903112/?query=user.id%3A21d8936b8bd24158&referrer=issue-stream&statsPeriod=14d&stream_index=0

adeebshihadeh, 2025-02-22:
@deanlee want to check this out?

deanlee, 2025-02-22:
It seems the crash occurred after the driver camera dropped requests.

> camera 2 dropped requests 19 2
> camera 2 dropped requests 19 2

deanlee, 2025-02-22:
The sequence of events leading to the crash is as follows:
1. A timeout in CAM_SYNC_WAIT triggers enqueue_buffer to call clear_req_queue().
2. This call to clear_req_queue() causes handle_camera_event to detect a camera_realign and invoke clear_req_queue() again.
3. The check for dropped requests returns true and calls enqueue_req_multi(), which overwrites the buf_idx of the current request_id, creating an invalid sync object.
4.  In subsequent enqueue_buffer calls, the function waits on this invalid sync object, causing another timeout and triggering another call to clear_req_queue().
5. The system then logs the error: "cam_context_prepare_dev_to_hw: 307 [cam-icp][1] No more request obj free."
6. This leads to an assertion failure and a crash.

Solutions:

1. Improve handling of dropped requests,  avoiding overwriting buf_idx and the creation of invalid sync objects after camera realigned. https://github.com/commaai/openpilot/pull/34665
2. Review error handling for CAM_SYNC_WAIT timeout, Calling clear_req_queue() immediately after a CAM_SYNC_WAIT timeout may be too aggressive. A more cautious and reliable error handling strategy should be considered.

deanlee, 2025-02-22:
> [2.286699] MAIN 0 kernel - CAM_ERR: CAM-CTXT: cam_context_prepare_dev_to_hw: 307 [cam-icp][1] No more request obj free
> [2.286758] MAIN 0 kernel - CAM_ERR: CAM-ICP: __cam_icp_config_dev_in_ready: 129 Failed to prepare device
> [2.286815] MAIN 0 kernel - CAM_ERR: CAM-CORE: __cam_node_handle_config_dev: 243 Config failure for node cam-icp
> [2.286878] MAIN 0 kernel - CAM_ERR: CAM-CORE: cam_node_handle_ioctl: 613 config device failed(rc = -12)

`config device failed(rc = -12)` rc = -12 corresponds to ENOMEM (Out of memory)

closed 2025-03-03 by adeebshihadeh