commaai/openpilot #34666: Bug: 0.9.8 logs are unreadable with useradmin website nor plotjuggler.

mvl-boston, 2025-02-22:
### Describe the bug

All my 0.9.8-based logs are unreadable with useradmin website nor plotjuggler.  My logs were from both a master and nightly based build from earlier this week (both in attempts to debug and get a test route for a planned pr for an openDBC change).

All my logs from runs with 0.9.7-based versions (mostly openpilot devel branch) work fine in both of the above.

On useradmin, when I click any segment, the log dropdown is empty, even though I see logs on the screen that shows the segments.

On plotjuggler, I get below:

(openpilot) mvl@[machineNameHidden]:~/openpilot/tools/plotjuggler$ ./juggle.py "ad9840558640c31d/00000000--7bd550bd35/0"
  0%|                                                                                             | 0/1 [00:01<?, ?it/s]
Traceback (most recent call last):
  File "/home/mvl/openpilot/tools/plotjuggler/./juggle.py", line 134, in <module>
    juggle_route(route_or_segment_name, args.can, args.layout, args.dbc)
  File "/home/mvl/openpilot/tools/plotjuggler/./juggle.py", line 82, in juggle_route
    all_data = sr.run_across_segments(24, partial(process, can))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mvl/openpilot/openpilot/tools/lib/logreader.py", line 298, in run_across_segments
    for p in tqdm.tqdm(pool.imap(partial(self._run_on_segment, func), range(num_segs)), total=num_segs, desc=desc):
  File "/home/mvl/openpilot/.venv/lib/python3.12/site-packages/tqdm/std.py", line 1181, in __iter__
    for obj in iterable:
  File "/usr/lib/python3.12/multiprocessing/pool.py", line 873, in next
    raise value
multiprocessing.pool.MaybeEncodingError: Error sending result: '<multiprocessing.pool.ExceptionWithTraceback object at 0x7f89bc9b02f0>'. Reason: 'PicklingError("Can't pickle <class 'zstd.ZstdError'>: import of module 'zstd' failed")'

### OS Version

Ubuntu 24.04 for windows 10

### openpilot version or commit

not sure how to find this version number on my computer.

sshane, 2025-02-22:
I fixed the second issue by updating my dependencies with `uv sync`

sshane, 2025-02-27:
Try to reinstall the uv venv

closed 2025-02-27 by sshane