commaai/openpilot #34688: `cereal`: `carParams` returns `ServerError` on `getMessage`

BBBmau, 2025-02-23:
### Describe the bug

when attempting to get the car branch by using `getMessage{"service": "carParams"}` I get the following response after timeout:

[image]

Assumed that it was due to the car not being turned on, however I sshed into the device while car was in ignition and found the following while still getting a `ServerError` with `getMessage`

[image]

### Provide a route where the issue occurs

n/a

### openpilot version

any

incognitojam, 2025-02-23:
Is `carParams` published by `card` which only runs when onroad?

BBBmau, 2025-02-23:
> Is `carParams` published by `card` which only runs when onroad? `athenad.getMessage` don't read params it waits for a message from running services

I tested this also when device was in the onRoad state and still received the ServerError message.

incognitojam, 2025-02-23:
It might have just timed out because the message is only sent every 50 seconds? https://github.com/commaai/openpilot/blob/master/selfdrive/car/card.py#L208

closed 2025-02-27 by sshane