commaai/openpilot #34702: Clip `carControl` outputs to ISO11270/15622 limits

adeebshihadeh, 2025-02-24:


jyoung8607, 2025-03-04:
Resolved by #34651?

sshane, 2025-03-04:
Lateral is done. Tesla route where model curvature > desired curvature due to the clip, we were achieving the desired curvature, and it threw the expected saturated alert: https://connect.comma.ai/2c912ca5de3b1ee9/00000106--cafd8c31c9/99/125

[image]

sshane, 2025-03-04:
Honda and GM request >350 brake units, but we believe that the brake pressure units don't fully correspond to longitudinal acceleration. From the comment in the values file:

```
  # Allow small margin below -3.5 m/s^2 from ISO 15622:2018 since we
  # perform the closed loop control, and might need some
  # to apply some more braking if we're on a downhill slope.
  # Our controller should still keep the 2 second average above
  # -3.5 m/s^2 as per planner limits
```

closed 2025-03-04 by sshane

sshane, 2025-03-04:
Tesla respects the lateral acceleration after roll compensation it looks like (which matches what we do), but it could apply 3.5 m/s^2 for about 0.5 seconds before it disengaged:

[image]