commaai/openpilot #34713: MacOS 15.3.1; unable to compile openpilot

dkiiv, 2025-02-26:
### Describe the bug

Hi,

For whatever reason, it looks like tinygrad isnt working in my openpilot instance?
I followed the steps in the tools/README, but have run into this issue each time

`Traceback (most recent call last):
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad_repo/examples/openpilot/compile3.py", line 136, in <module>
    test_val = compile(onnx_file) if not getenv("RUN") else None
               ^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad_repo/examples/openpilot/compile3.py", line 31, in compile
    new_inputs = {k:Tensor.randn(*shp, dtype=_from_np_dtype(input_types[k])).mul(8).realize() for k,shp in sorted(input_shapes.items())}
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad_repo/examples/openpilot/compile3.py", line 31, in <dictcomp>
    new_inputs = {k:Tensor.randn(*shp, dtype=_from_np_dtype(input_types[k])).mul(8).realize() for k,shp in sorted(input_shapes.items())}
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/tensor.py", line 4012, in _wrapper
    ret = fn(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/tensor.py", line 262, in realize
    run_schedule(*self.schedule_with_vars(*lst), do_update_stats=do_update_stats)
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 169, in run_schedule
    for ei in lower_schedule(schedule):
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 162, in lower_schedule
    raise e
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 156, in lower_schedule
    try: yield lower_schedule_item(si)
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 151, in lower_schedule_item
    def lower_schedule_item(si:ScheduleItem) -> ExecItem: return ExecItem(*cast(tuple[Runner,list], si_lowerer.rewrite(si.ast, si.bufs)), si.metadata)
                                                                                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/ops.py", line 825, in rewrite
    if (ret:=(fxn(ctx=ctx, **match) if has_ctx else fxn(**match))) is not None: return ret
              ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 145, in <lambda>
    (UPat(Ops.SINK, name="sink"), lambda ctx,sink: (runner:=get_runner(ctx[0].device, sink), [ctx[x] for x in runner.p.globals])),
                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 113, in get_runner
    prg: ProgramSpec = get_kernel(Device[device].renderer, ast).to_program()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/codegen/kernel.py", line 682, in to_program
    src = self.opts.render(self.uops)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/renderer/cstyle.py", line 157, in render
    l = cast(str, self.string_rewrite.rewrite(u, ctx=self))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/ops.py", line 825, in rewrite
    if (ret:=(fxn(ctx=ctx, **match) if has_ctx else fxn(**match))) is not None: return ret
              ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/renderer/cstyle.py", line 29, in <lambda>
    (UPat(Ops.SPECIAL, name="x"), lambda ctx,x: f"{ctx.code_for_workitem[x.arg[0][0]](x.arg[0][-1])}; /* {x.arg[1]} */"),
                                                   ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
KeyError: 'i'
scons: *** [selfdrive/modeld/models/supercombo_tinygrad.pkl] Error 1
scons: building terminated because of errors.`

### OS Version

Sequoia 15.3.1

### openpilot version or commit

a8c14a4e10b24c3bc6cadc5af86e3db61d4a0e96

dkiiv, 2025-02-26:
I went back 2 weeks and chose a random commit;
`efda24dbe156fff4ee3a14bb9c9b289d4666a3b5`

now it compiles and works after following the instructions in the README.. so something definitely broke in newer tinygrad (i saw tiny grad update when i did git submodule update after checking out to the commit)

dkiiv, 2025-03-05:
if i find the time in a couple days, i can do the ``git bisect`` routine to find the specific commit which broke it for mac users
in the mean time, @Aniurm if you'd like to, i posted the commit which works.. knowing and testing against HEAD should let you also find the problem commit

dkiiv, 2025-03-21:
Apple Silicon(ARM), m1 pro

dkiiv, 2025-03-21:
maybe whatever broke on the commit mentioned in the issue was fixed later on.

does it still compile/work for you even on the commit i have posted in the issue?

BBBmau, 2025-03-27:
> maybe whatever broke on the commit mentioned in the issue was fixed later on.
> 
> does it still compile/work for you even on the commit i have posted in the issue?

yep just went and built openpilot and was able to run the ui on my mac just now. I even just recently upgraded to 15.3.2 as well

adeebshihadeh, 2025-05-17:
still an issue after bumping tinygrad?

dkiiv, 2025-05-17:
sadly, yes.
this is after doing git pull, updating submodules, re-running the mac_setup.sh script, and then running the scons command in the OP environment

```
Traceback (most recent call last):
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad_repo/examples/openpilot/compile3.py", line 136, in <module>
    test_val = compile(onnx_file) if not getenv("RUN") else None
               ^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad_repo/examples/openpilot/compile3.py", line 30, in compile
    new_inputs = {k:Tensor.randn(*shp, dtype=_from_np_dtype(input_types[k])).mul(8).realize() for k,shp in sorted(input_shapes.items())}
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad_repo/examples/openpilot/compile3.py", line 30, in <dictcomp>
    new_inputs = {k:Tensor.randn(*shp, dtype=_from_np_dtype(input_types[k])).mul(8).realize() for k,shp in sorted(input_shapes.items())}
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/tensor.py", line 4245, in _wrapper
    ret = fn(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/tensor.py", line 270, in realize
    run_schedule(*self.schedule_with_vars(*lst), do_update_stats=do_update_stats)
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 170, in run_schedule
    for si, ei in lower_schedule(schedule):
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 163, in lower_schedule
    raise e
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 157, in lower_schedule
    try: yield (si, lower_schedule_item(si))
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 152, in lower_schedule_item
    return ExecItem(*cast(tuple[Runner,list], si_lowerer.rewrite(si.ast, si.bufs)), si.metadata, si.fixedvars)
                                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/ops.py", line 876, in rewrite
    if (ret:=match(uop, ctx)) is not None: return ret
             ^^^^^^^^^^^^^^^
  File "<string>", line 3, in compiled_match
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 145, in <lambda>
    (UPat(Ops.SINK, name="sink"), lambda ctx,sink: (runner:=get_runner(ctx[0].device, sink), [ctx[x] for x in runner.p.globals])),
                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/engine/realize.py", line 112, in get_runner
    prg: ProgramSpec = get_kernel(Device[device].renderer, ast).to_program()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/codegen/kernel.py", line 568, in to_program
    src = self.opts.render(self.uops)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/renderer/cstyle.py", line 177, in render
    def render(self, uops:list[UOp]) -> str: return self.render_kernel(*self._render(uops), uops)
                                                                        ^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/renderer/cstyle.py", line 158, in _render
    l = cast(str, self.string_rewrite.rewrite(u, ctx=self))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/ops.py", line 876, in rewrite
    if (ret:=match(uop, ctx)) is not None: return ret
             ^^^^^^^^^^^^^^^
  File "<string>", line 3, in compiled_match
  File "/Users/khonsu/Github/commaTools/openpilot/tinygrad/renderer/cstyle.py", line 29, in <lambda>
    (UPat(Ops.SPECIAL, name="x"), lambda ctx,x: f"{ctx.code_for_workitem[x.arg[0][0]](x.arg[0][-1])}; /* {x.arg[1]} */"),
                                                   ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
KeyError: 'i'
scons: *** [selfdrive/modeld/models/driving_vision_tinygrad.pkl] Error 1
scons: *** [selfdrive/modeld/models/dmonitoring_model_tinygrad.pkl] Error 1
Warning: Please note that the following versions of CasADi  are officially supported: 3.5.6 or 3.5.5 or 3.5.4 or 3.5.3 or 3.5.2 or 3.5.1 or 3.4.5 or 3.4.0.
 If there is an incompatibility with the CasADi generated code, please consider changing your CasADi version.
Version 3.7.0 currently in use.
scons: building terminated because of errors.
```

adeebshihadeh, 2025-05-17:
Probably time to make an issue with tinygrad then

dkiiv, 2025-05-17:
providing i know how to replicate it on the tinygrad repo, sure. but i'm not sure how to do that.

unless i just use this compiling openpilot as the source in the issue on that repo

andiradulescu, 2025-05-18:
I already hit this but with METAL=1 and submitted a [tinygrad issue](https://github.com/tinygrad/tinygrad/issues/10169) and also got a reply.

I think for some reason the compile (in your case) is done (by default) with METAL=1 (but not with IMAGE=0 NOLOCALS=0).

Try replacing CLANG=1 with CPU=1
https://github.com/commaai/openpilot/blob/33721cf5fab014f945df783fc3267f4081c23401/selfdrive/modeld/SConscript#L49
since CLANG was replaced with CPU in tinygrad.

adeebshihadeh, 2025-05-18:
Merged that PR. All good now?

dkiiv, 2025-05-18:
all good! it compiles after that merge!

```
scons: done building targets.
```

f630cac06

closed 2025-05-18 by dkiiv