commaai/openpilot #34721: `dmonitoringmodeld` memory leak

maxime-desroches, 2025-02-27:
`dmonitoringmodeld` get killed by `lowmemorykiller` when using ~1.4GB of memory after ~10 hours on 8 devices.

The memory usage increases by exactly 1024 kB every ~1 minutes.

royjr, 2025-03-02:
Not sure if its related, but the ui freezes at times. SSH is fine, htop never loads, but top does. `op start` doesn't help
[image]
```[ 4121.584092]   pwq 16: cpus=0-7 flags=0x4 nice=0 active=1/1
[ 4121.584099]     in-flight: 326:socket_read_work_fn
[ 4121.584107] workqueue DIAG_SOCKMODEM_DCI_CMD: flags=0xe000a
[ 4121.584110]   pwq 16: cpus=0-7 flags=0x4 nice=0 active=1/1
[ 4121.584114]     in-flight: 84:socket_read_work_fn
[ 4121.584121] workqueue DIAG_SOCKLPASS_CNTL: flags=0xe000a
[ 4121.584124]   pwq 16: cpus=0-7 flags=0x4 nice=0 active=1/1
[ 4121.584128]     in-flight: 353:socket_read_work_fn
[ 4121.584137] workqueue DIAG_SOCKCDSP_CNTL: flags=0xe000a
[ 4121.584140]   pwq 16: cpus=0-7 flags=0x4 nice=0 active=1/1
[ 4121.584143]     in-flight: 354:socket_read_work_fn
[ 4121.584166] workqueue crm_workq-icp_timer_que: flags=0x50
[ 4121.584170]   pwq 13: cpus=6 node=0 flags=0x0 nice=-20 active=1/256
[ 4121.584174]     pending: cam_req_mgr_process_workq
[ 4121.584201] workqueue crm_workq-980205-cc0109: flags=0x50
[ 4121.584205]   pwq 13: cpus=6 node=0 flags=0x0 nice=-20 active=1/1
[ 4121.584211]     pending: cam_req_mgr_process_workq
[ 4121.584221] workqueue crm_workq-dc020a-4010f: flags=0x50
[ 4121.584226]   pwq 13: cpus=6 node=0 flags=0x0 nice=-20 active=1/1
[ 4121.584231]     pending: cam_req_mgr_process_workq
[ 4121.584254] pool 16: cpus=0-7 flags=0x4 nice=0 hung=0s workers=8 idle: 60495 58463 59444 59858
[ 4123.309442] cam_tasklet_get_cmd: 91 callbacks suppressed
[ 4123.309443] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.309451] cam_irq_controller_th_processing: 91 callbacks suppressed
[ 4123.309452] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.359440] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.359445] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.409442] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.409446] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.459445] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.459448] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.509443] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.509447] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.559448] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.559452] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.609446] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.609450] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.659451] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.659455] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.709449] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.709453] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload
[ 4123.759453] CAM_ERR: CAM-ISP: cam_tasklet_get_cmd: 103 No more free tasklet cmd
[ 4123.759457] CAM_ERR: CAM-ISP: cam_irq_controller_th_processing: 620 Can't get bh payload```

adeebshihadeh, 2025-03-06:
It's this https://github.com/tinygrad/tinygrad/issues/9351.

@haraschax mitigated this by reverting to an older tinygrad that leaks 10x less, but definitely would like to fix all the leaks before 0.9.9.

adeebshihadeh, 2025-04-25:
@maxime-desroches let's try latest tinygrad? the last issue is still open, but says it should be fixed

adeebshihadeh, 2025-04-30:
@haraschax looks like we still have a tinygrad fork. what's the process for bumping? any reason we still need a fork?

closed 2025-05-06 by adeebshihadeh