commaai/openpilot #34722: Cabana bugs and improvements

sshane, 2025-02-27:
- [ ] --zmq `ip` doesn't autofill it, you have to type it twice
- [ ] scrolling doesn't seem smooth, it ignores scrolling input

@deanlee

deanlee, 2025-02-28:
> scrolling doesn't seem smooth, it ignores ~half of your fast scrolling input

We don't handle scroll events directly in Cabana. The issue is likely due to Qt's default behavior of compressing scroll events, especially during fast scrolling, which can lead to skipped inputs. Additionally, this behavior may vary depending on the platform or input device.

sshane, 2025-03-07:
@deanlee it would be much appreciated if Cabana could use LogReader's auto sourcing abilities. So many old routes and the ones in opendbc's routes.py do not work in Cabana

deanlee, 2025-03-14:
> Maybe a ./cabana.py Python wrapper that finds the URLs then passes into Cabana? So you don't have to re-write it in C++

Modified logreader.py to optionally output log identifiers when --identifiers-only is specified. This list is then used to load corresponding logs into Cabana. See details in the pull request:

https://github.com/commaai/openpilot/pull/34863

sshane, 2025-03-27:
It would be nice to have a reset button that goes back to the latest datapoints in the stream after you've paused and zoomed. It's difficult to click constantly right at the end to skip back to the latest data.

deanlee, 2025-04-09:
> It would be nice to have a reset button that goes back to the latest datapoints in the stream after you've paused and zoomed. It's difficult to click constantly right at the end to skip back to the latest data.

Cabana has already implemented a 'Skip to the End' button, which moves the cursor to the most recent data points. Does this meet your requirements?:

[image]

sshane, 2026-03-14:
cc @adeebshihadeh

closed 2026-03-14 by adeebshihadeh