commaai/openpilot #34734: comma 3X: switch to kernel driver for low level SPI comms

adeebshihadeh, 2025-02-28:
The comma 3X talks to its internal panda over SPI using the [spidev](https://www.kernel.org/doc/Documentation/spi/spidev) driver. We want to move the low-level SPI transactions into the kernel 

I already started this at [panda/drivers/spi/](https://github.com/commaai/panda/tree/master/drivers/spi).

Minimum requirements for this bounty:
* Make it interrupt driven (no busy waiting!)
* Good auto loading + updating the driver
* Better perf than the existing panda SPI stack

We'll be switching `pandad` in openpilot to Python soon, so no need to implement this for the C++ panda library (just the one in `panda/python/`).

andiradulescu, 2025-03-06:
Unfortunately you can't, C3 has the panda connected via USB:
```
$ lsusb
Bus 001 Device 010: ID 3801:ddcc comma.ai panda
```

robbederks, 2025-03-13:
> You need a C3X, C3 has the panda connected via USB:
> 
> ```
> $ lsusb
> Bus 001 Device 010: ID 3801:ddcc comma.ai panda
> ```

It's also connected over SPI actually, if you compile it with `ENABLE_SPI=1` it should also be possible on tici

robin-reckmann, 2025-04-08:
@robbederks Yep, I could confirm that SPI works on my C3, but only for frequencies below 33MHz.

jakethesnake420, 2025-08-06:
Thinking about doing this. 

- [ ]  Kernel SPI protocol driver that handles low level async communication.
- [ ]  Update panda spi userspace code to eliminate the need for busy looping

glfldh, 2026-02-27:
Hi @adeebshihadeh,

I am very interested in the kernel SPI driver bounty (#34734) for comma 3X.

**My Background:**
- Experience with Linux kernel driver development
- Familiar with SPI protocol and interrupt-driven I/O
- Prior work on embedded systems performance optimization

**Understanding the Requirements:**
- Replace spidev with custom kernel driver for panda SPI communication
- Implement interrupt-driven approach for higher throughput
- Target: Significant performance improvement over current spidev polling
- Bonus: +$250 for pandad Python rewrite

**Technical Questions:**

1. **Hardware Access**: Do I need physical comma 3X hardware for development? Or can I test via emulation/remote access?

2. **Performance Baseline**: What is the current SPI throughput with spidev? What target throughput should the kernel driver achieve?

3. **Driver Architecture**: Should this be a full Linux kernel module, or a userspace driver with kernel-bypass (like DPDK-style)?

4. **Integration**: How does the driver integrate with existing panda communication stack? Need to modify panda firmware or just host-side?

5. **Testing**: Are there specific benchmarks or stress tests I should run to validate the throughput improvement?

**Proposed Approach:**
- Phase 1: Kernel module skeleton + basic SPI register access
- Phase 2: Interrupt implementation + DMA support if applicable
- Phase 3: Performance optimization to hit throughput targets
- Phase 4 (Bonus): pandad Python to C++ migration

Could you confirm the $750-1000 bounty range and any specific timeline expectations? I can start immediately.

@glfldh

glfldh, 2026-02-27:
Hi @adeebshihadeh, I am very interested in the kernel SPI driver bounty (#34734) for comma 3X. I have experience with Linux kernel driver development and SPI protocol. Could you confirm the $750-1000 bounty range and assign me to this task? I can start immediately. @glfldh