commaai/openpilot #34742: `release3-staging` Error message: Process not running (loggerd)

yuzisee, 2025-03-01:
### Describe the bug

I got "Process not running" (loggerd) during `release3-staging`

But it won't create logs to upload if loggerd isn't working... so I don't have a route to share? (At least, connect.comma.ai isn't showing this particular route)

### Provide a route where the issue occurs

Dongle ID is 21015c36062c3cef (without loggerd working there is no logs, right?)

### openpilot version

release3-staging (installed Feb 28, 2025)

### Additional info

Original thread: https://discord.com/channels/469524606043160576/616456819027607567/1345445493869777119

[image]

adeebshihadeh, 2025-03-01:
https://commaai.sentry.io/issues/5994831372/?query=user.id%3A21015c36062c3cef&referrer=issue-stream&statsPeriod=14d&stream_index=0

yuzisee, 2025-03-01:
> It hit [this assert](https://github.com/commaai/openpilot/blob/2d882b716add9b2316760d7e5425cf5e35391b21/system/loggerd/logger.cc#L181). 

Hmm, looks like it's trying to create a folder for the route itself?

> Do you ever SSH into your device?

Once long ago, but not on this device in over a year at least and haven't had any issues since.

> Is it possible you changed the perms on the log directory? Was it fine on subsequent drives?

Previous drives also work (I can see routes I drove earlier that same morning with `release3-staging` and they successfully uploaded and are visible on connect.comma.ai

Also, let me see if Settings > Device > Reboot fixes the  issue... stay tuned.
 
> https://commaai.sentry.io/issues/5994831372/?query=user.id%3A21015c36062c3cef&referrer=issue-stream&statsPeriod=14d&stream_index=0

yuzisee, 2025-03-01:
> Also, let me see if Settings > Device > Reboot fixes the issue... stay tuned.

Yes. Rebooting the device also fixes it... so it couldn't be a permissions issue right?

No reason for the permission to change
- (a) by switching vehicle ignition from **on** → **off** → **on**
- nor, (b) by rebooting the openpilot device
right?

adeebshihadeh, 2025-03-01:
Ah, if you haven't SSH'd in for a long time or used a fork without doing a factory reset, then this should definitely be handled.

yuzisee, 2025-03-01:
> Ah, if you haven't SSH'd in for a long time or used a fork without doing a factory reset, then this should definitely be handled.

But even https://github.com/commaai/openpilot/commit/9735cf2eb6089bfa6e548d57aef1750055d9991f already shipped in https://github.com/commaai/openpilot/releases/tag/v0.9.7 so if this is, in fact, a 0.9.8 regression it would have to
1. be some sort of timing issue / race condition?
2. maybe a buffer overflow (the path itself is corrupted at some point)

For what it's worth, the **last** path to upload correctly before the issue was
`00000000--7b39adbf9f`
and then the **first** path to upload correctly (after the reboot) was
`00000005--73586bbde4`

so presumably I attempted to engage openpilot 4 times in between (sounds about right) and all of those hit the assert (you should see exactly 4 reports of that assert triggering from my device, if so)

adeebshihadeh, 2025-03-01:
So that number is the route count, not engaged count (route = one "ignition" session). If this is a regression, I suspect it has to do with an NVMe race condition. If that's true, I'd expect the issue to start on the first route after a bootup and persist until reboot.

adeebshihadeh, 2025-03-01:
Yeah, it's NVMe related. You're also getting:
```python
Traceback (most recent call last):
  File "/data/openpilot/system/loggerd/uploader.py", line 58, in listdir_by_creation
    paths = [f for f in os.listdir(d) if os.path.isdir(os.path.join(d, f))]
                        ^^^^^^^^^^^^^
OSError: [Errno 5] Input/output error: '/data/media/0/realdata/'
```

yuzisee, 2025-03-03:
> If it happens again, it'd be great if you can post the output of this:
> 
> ```
> dmesg | nc termbin.com 9999
> journalctl -xe | nc termbin.com 9999
> ```

FWIW it's happened in four separate occasions since installed `release3-staging` and it _seems_ to be easiest to reproduce by just leaving the device on for a long time after a drive or two.

Here are the kernel & systemd logs:

[dmesg.txt](https://github.com/user-attachments/files/19047843/dmesg.txt)

[journalctl_xe.txt](https://github.com/user-attachments/files/19047842/journalctl_xe.txt)

adeebshihadeh, 2025-03-03:
Your NVMe is getting disconnected. 
```
[30106.651116] nvme 0001:01:00.0: Failed status: 0xffffffff, reset controller.
[30106.651282] nvme 0001:01:00.0: of_irq_parse_pci() failed with rc=134
[30106.656727] q6asm_callback: cmd = 0x10bcd returned error = 0x1
[30106.656805] __q6asm_cmd: DSP returned error[ADSP_EFAILED] opcode 68557
[30106.671085] pci_raw_set_power_state: 20 callbacks suppressed
[30106.671094] nvme 0001:01:00.0: Refused to change power state, currently in D3
[30106.671153] nvme nvme0: Removing after probe failure status: -19
[30106.671433] nvme0n1: detected capacity change from 250059350016 to 0
```

Although we made big userspace updates, the kernel is largely unchanged. It's possible this is a HW failure rather than a regression in the software. Since it's so reproducible, can you try to repro on the current release?

adeebshihadeh, 2025-03-03:
It seemed to happen right as you went onroad though... perhaps it's reproducible by cycling ignition rather than pure uptime.

adeebshihadeh, 2025-03-03:
Ah, I forgot that I recently made this stricter: https://github.com/commaai/openpilot/commit/82722217be3ab050945c0240224ddcbab565f412.

The NVMe behavior likely always happened on your device, but it didn't show the alert until now. I'm going to just revert that for this release and revisit for the next one.

yuzisee, 2025-03-03:
> It seemed to happen right as you went onroad though... perhaps it's reproducible by cycling ignition rather than pure uptime.

Yeah. I did see it only once in the middle of driving (it would have been the `00000012-`… route, which never uploaded for the same reason) but the other times were just as I went onroad, that's true. However, it's possible that the NVME issue occurs _while parked_ and the assertion triggers only once we go onroad (because there's no reason for https://github.com/commaai/openpilot/commit/82722217be3ab050945c0240224ddcbab565f412 to trigger when parked)

> Since it's so reproducible, can you try to repro on the current release?

Once on release, is there anything else I can log or capture other than `dmesg` and `journalctl` that would be helpful?

> I'm going to just revert that for this release and revisit for the next one.

If we revert the change, I won't be able to tell when the issue triggers — so I won't know when to `dmesg`/`journalctl` I assume?

adeebshihadeh, 2025-04-04:
Made it strict again, except for C3 with NVMe - https://github.com/commaai/openpilot/pull/34978

yuzisee, 2025-05-30:
Happening again on release3-staging (@adeebshihadeh not sure if you want to re-open this issue or create a new one)
[image]