commaai/openpilot #34746: Exception: setup_quectel failed after retry

royjr, 2025-03-02:
### Describe the bug

This is running off https://github.com/commaai/opendbc/pull/1575, but I've been seeing these errors for a year now and can't seem to figure it out. Similar to https://github.com/commaai/openpilot/issues/30208, https://github.com/commaai/openpilot/issues/33996, https://github.com/commaai/openpilot/pull/28952

<details>
<summary>tmux logs with debugging</summary>

```
waiting for weston
weston ready after 0s
ln: failed to create symbolic link '/home/comma/.cursor-server/vscode-server': File exists
/data/openpilot has been modified, skipping overlay update installation
Using Wayland-EGL
MESA: error: ZINK: vkCreateInstance failed (VK_ERROR_INCOMPATIBLE_DRIVER)
libEGL warning: egl: failed to create dri2 screen
qt.qpa.wayland: "wl-shell" is a deprecated shell extension, prefer using "xdg-shell-v6" or "xdg-shell" if supported by the compositor by setting the environment variable QT_WAYLAND_SHELL_INTEGRATION
Using the 'wl-shell' shell integration
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Retrieved `panda/board/jungle/main-panda_jungle.o' from cache
Retrieved `panda/board/jungle/main-panda_jungle_h7.o' from cache
Retrieved `panda/board/main-panda.o' from cache
Retrieved `panda/board/jungle/obj/panda_jungle_h7.elf' from cache
Retrieved `panda/board/main-panda_h7.o' from cache
Retrieved `panda/board/jungle/obj/panda_jungle_h7.bin' from cache
Retrieved `panda/board/jungle/obj/panda_jungle_h7.bin.signed' from cache
Retrieved `panda/board/obj/panda_h7.elf' from cache
Retrieved `panda/board/obj/panda_h7.bin' from cache
Retrieved `panda/board/obj/panda_h7.bin.signed' from cache
scons: done building targets.
system/loggerd/bootlog.cc: bootlog to /data/media/0/realdata/boot/000000e6--07dd0fb854.zst
gbm_create_device(156): Info: backend name is: msm_drm
updates are disabled by the DisableUpdates param
Panda 35003c000a51333038363231 connected, version: DEV-2c802449-DEBU, signature b7f6738fee2b4436, expected 620b36b3081a6d6c
flash: unlocking
INFO:panda:flash: unlocking
flash: erasing sectors 1 - 1
INFO:panda:flash: erasing sectors 1 - 1
configuring modem
Successfully set initial EPS bearer properties
selfdrive/ui/qt/network/wifi_manager.cc: Changing gsm.auto-config to true
logmessaged timed ui deleter pandad hardwared tombstoned updated uploader statsd
: "wl-shell" is a deprecated shell extension, prefer using "xdg-shell-v6" or "xdg-shell" if supported by the compositor by setting the environment variable QT_WAYLAND_SHELL_INTEGRATION
flash: flashing
INFO:panda:flash: flashing
logmessaged timed ui deleter pandad hardwared tombstoned updated uploader statsd
pid 321's current affinity list: 3
pid 321's new affinity list: 3
logmessaged timed ui deleter pandad hardwared tombstoned updated uploader statsd
logmessaged timed ui deleter pandad hardwared tombstoned updated uploader statsd
logmessaged timed ui deleter pandad hardwared tombstoned updated uploader statsd
logmessaged timed ui deleter pandad hardwared tombstoned updated uploader statsd
flash: resetting
INFO:panda:flash: resetting
logmessaged timed ui deleter pandad hardwared tombstoned updated uploader statsd
selfdrive/pandad/main.cc: starting pandad
selfdrive/pandad/pandad.cc: connecting to pandas: 35003c000a51333038363231
selfdrive/pandad/panda.cc: connected to 35003c000a51333038363231 over SPI
selfdrive/pandad/pandad.cc: connected to all pandas
system/camerad/cameras/spectra.cc: VIDIOC_CAM_CONTROL error: op_code 266 - errno 19
Waiting for CAN messages...
waiting for modem to come up
Getting VIN & FW versions
Getting VIN & FW versions
Setting OBD multiplexing to True
OBD multiplexing set successfully
system/camerad/cameras/spectra.cc: VIDIOC_CAM_CONTROL error: op_code 266 - errno 19
Unexpected opcode: 39, Payload: fd1b010000000000000100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
00000000000000000000000000000000000000000
[... 676 lines cut ...]
  -----------------------------------
  3GPP     |                    imei: 101010101010101
           |           enabled locks: fixed-dialing
           |             operator id: 310260
           |           operator name: T-Mobile
           |            registration: home
           |    packet service state: attached
  -----------------------------------
  3GPP EPS |    ue mode of operation: csps-2
           |     initial bearer path: /org/freedesktop/ModemManager1/Bearer/0
           |  initial bearer ip type: ipv4
  -----------------------------------
  SIM      |        primary sim path: /org/freedesktop/ModemManager1/SIM/0
  -----------------------------------
  Bearer   |                   paths: /org/freedesktop/ModemManager1/Bearer/1
```
</details>

### Provide a route where the issue occurs

e6fd6ef0e9cc1171/00000197--5d9d371e2c

### openpilot version

master

adeebshihadeh, 2025-03-20:
I don't see anyone hitting this in release. Are you using external storage or something?

royjr, 2025-03-20:
I'm holding off on playing with external storage for now. Normally see it while making local port changes and using op start. Recently started fresh, I'll report back if I see it again.

royjr, 2025-03-22:
Just happened again. Clear skys, no usb/joy/aux. Modified files through ssh, op start, and same error.
`e6fd6ef0e9cc1171/0000008e--6b282424ad`

royjr, 2025-04-13:
Confirmed by another user, it sometimes happens when using op start and not rebooting the device. Maybe qcomgpsd doesn't like the way op is restarted.