commaai/openpilot #34787: updater broken

royjr, 2025-03-06:
### Describe the bug

can't really point to anything specific, but (on a port) updater checks for updates, downloads, installs, reboots and goes back to the same version (ie. doesn't update) and only seems to have started a few days ago, not sure if its agnos related. https://discord.com/channels/469524606043160576/1317666784446255135/1346904362919919668

Another user having the same issue on another port. https://discord.com/channels/469524606043160576/534359517836607488/1347044708362227752

### Provide a route where the issue occurs

master

### openpilot version

master

haram-KONA, 2025-03-07:
I am experiencing the same issue repeatedly. It keeps repeating the update (automatically) but the actual update does not occur and it keeps repeating the update again.

haram-KONA, 2025-03-07:
I think that forked branches are experiencing this phenomenon. After installing from the original Comma site, the update error does not occur.

LouisVelez, 2025-03-08:
Im on nightly-dev and also experiencing this issue. Stuck since March2 update.

RGBacon, 2025-03-10:
> Im on nightly-dev and also experiencing this issue. Stuck since March2 update.

Same. I was originally stuck on nightly, and then switched to nightly-dev, but the issue persisted.

jayvin100, 2025-03-10:
Tesla Model 3 HW3 here, using nightly-dev also facing the same issue. My workaround is reinstalling the branch every time there's an update.

adeebshihadeh, 2025-03-10:
Fixed by https://github.com/commaai/openpilot/commit/f870e48b3a13e72b5da20beec56c6be171f1618b

closed 2025-03-10 by adeebshihadeh