commaai/openpilot #34788: op.sh: op start triggers reset (sometimes)

royjr, 2025-03-06:
### Describe the bug

op start is much more successful after https://github.com/commaai/agnos-builder/commit/d39db39e772de4bd700193403f8c4c5f09eb94cd but still triggers the reset screen once (per boot maybe) and works correctly once dismissed once

### Provide a route where the issue occurs

master

### openpilot version

master

adeebshihadeh, 2025-03-07:
Need a repro

royjr, 2025-03-11:
Not exactly a repro, but just tracking what I remember before it happened today. Its happens once (not sure why), then fine for the rest of the "session"

- car is on and in park
- comma connected to phone hotspot and using local ssh
- toggled op long on and off a few times before this happened
- modified a file (while working on a port)
- op start.. FAILS
```
waiting for weston
weston ready after 0s
launching system reset, got taps
Using Wayland-EGL
qt.qpa.wayland: "wl-shell" is a deprecated shell extension, prefer using "xdg-shell-v6" or "xdg-shell" if supported by the compositor by setting the environment variable QT_WAYLAND_SHELL_INTEGRATION
Using the 'wl-shell' shell integration
```
- checked /tmp/booted.. it exists! (checked after got taps)
- checked touch_count
```
cat /sys/devices/platform/soc/894000.i2c/i2c-2/2-0017/touch_count
31
```
- op start.. SUCCEEDS (without touching the screen)

royjr, 2025-03-23:
just got `got taps` with a black screen, a second `op start` fixed it

royjr, 2025-03-23:
[image]

royjr, 2025-08-07:
Still happening one out of three times an `op start` is done, a second `op start` always fixes it. Another user reports same.