commaai/openpilot #34817: Camera Malfunction: roadCamera, wideRoadCamera

andiradulescu, 2025-03-07:
### Describe the bug

I got this on C3X after turning ignition on. After cycling ignition off and on, worked fine (w/o restarting the C3X). DM camera was off (Record and Upload Driver Camera toggled OFF).

[0000014f--7cf25f25e5--0.zip](https://github.com/user-attachments/files/19134408/0000014f--7cf25f25e5--0.zip)

### Provide a route where the issue occurs

0000014f--7cf25f25e5--0

### openpilot version

release3-staging / 4e4f5ab99839f5b991686d548669f4280072d4c1

### Additional info

A glimpse from the logs:
```
{"filename": "system/camerad/cameras/spectra.cc", "funcname": "clearAndRequeue", "levelnum": 30, "lineno": 1392, "msg": "clearing and requeuing camera 2 from 6"}
{"filename": "system/camerad/cameras/spectra.cc", "funcname": "do_sync_control", "levelnum": 40, "lineno": 60, "msg": "CAM_SYNC error: id 6 - errno 11 - ret 0 - ioctl_result -110"}

CAM_WARN: CAM-CRM: cam_req_mgr_process_add_req: 1708 Unexpected state 2 for slot 17 map 1

{"filename": "system/camerad/cameras/spectra.cc", "funcname": "handle_camera_event", "levelnum": 40, "lineno": 1348, "msg": "camera 2 sync failure 7 10 "}
{"filename": "system/camerad/cameras/spectra.cc", "funcname": "validateEvent", "levelnum": 40, "lineno": 1376, "msg": "camera 0 frame ID skipped, 6 -> 12"}
{"filename": "system/camerad/cameras/spectra.cc", "funcname": "validateEvent", "levelnum": 40, "lineno": 1376, "msg": "camera 1 frame ID skipped, 6 -> 12"}
{"filename": "system/camerad/cameras/spectra.cc", "funcname": "do_sync_control", "levelnum": 40, "lineno": 60, "msg": "CAM_SYNC error: id 6 - errno 11 - ret 0 - ioctl_result -110"}

__q6asm_cmd: DSP returned error[ADSP_EFAILED] opcode 68557
(this repeats a lot of  times in the first segment)

CAM_ERR: CAM-SYNC: cam_sync_wait: 393 Error: timed out for sync obj = 37
CAM_WARN: CAM-CRM: cam_req_mgr_process_add_req: 1708 Unexpected state 2 for slot 0 map 1
```

royjr, 2025-03-08:
[image]

yuzisee, 2025-03-10:
I saw something similar in 21015c36062c3cef/0000002b--6d51b4db81 running `release3-staging` as well.

adeebshihadeh, 2025-03-10:
```
21015c36062c3cef/0000002b--6d51b4db81/0
- first BPS frame fails
- road cam missing for most of route
- 25 hours of uptime
```

adeebshihadeh, 2025-03-14:
Can't repro this at all with a `test_camera.py`/`test_onroad.py` loop or even the camerad "stress test". Last two cases:
```
6fea1470edf96da9/00000272--6026f90b05
559f9f7880533ed4/00000080--147080ab8e
```

---

Sequence goes like this:
1. First few dcam frames time out in the BPS (makes it out of the IFE before this)
2. Next dcam frame times out in the IFE
3. A few more dcam frame time outs, then it's fine
4. One or both of road and wide road will now skip request IDs for a few seconds
5. Everything is fine after this.

Either 1) the thing that's causing BPS issues messes up the other cameras' ISP pipelines or 2) it's the same root cause. #1 would be pretty bad and I haven't been able to repro by intentionally messing with the dcam requests, but I doubt many other people are using the ISP like this.

adeebshihadeh, 2025-03-14:
@deanlee it seems we can't rely on the fence timeout. want to add back per-camera threads but in a more minimal way? perhaps we can make the SOF polling per-thread?

royjr, 2025-03-14:
Drove fine, parked car, came back to this an hour later. Works normally after restart.
[image]

closed 2025-03-15 by adeebshihadeh

rahulkhadse, 2025-07-01:
Hi, I am still encountering the 'Camera Malfunction: roadCamera' error on my C3X, despite this issue being previously closed with a fix in #34871.

I have already performed the following troubleshooting steps:

- Reinstalled the software.
- Recalibrated the device.
- Reattached all wires securely.
- Tested the power cord with another Comma AI device to confirm it's functional.

My vehicle is a Honda Civic 2020. The error persists after these actions. An image of the error message on the device screen is attached.

Could you please reopen this issue for further investigation? I can provide new logs if needed - just let me know how can I avail those to you. 

[image]