commaai/openpilot #34836: Automated Firehose Mode uploads at the same priority in the queue as human-initiated interactive uploads

nelsonjchen, 2025-03-10:
As a comma connect user, I'll like my human-initiated "interactive" uploads prioritized over automated Firehose Mode uploads so it feels more responsive.

1. Have Firehose mode on (or I suspect just being on a branch of great interest to comma)
2. Automated Firehose mode segments are queued for upload.
3. Go to comma Connect, choose upload files or segments.
4. View Uploads for device, see the human chosen upload files or segments are queued behind automated firehose segments. This can be a 5-10 segment delay before the human chosen segments are finally uploaded.

Workaround: Cancel All Segments and requeue desired segments for upload.

adeebshihadeh, 2025-03-10:
Bounty is for setting up the priority-based upload in athena.

Must have a good unit test around it.

greatgitsby, 2025-03-11:
this can't be achieved w/o passing the "intent" or requester to `athenad` (`intent = Intent.COMMA_CONNECT | Intent.FIREHOSE`).

is this something that can be added on the connect / firehose service backends?

closed 2025-03-19 by incognitojam